PX4/PX4-user_guide #784: Doc Bug: Drotek Pixhawk 3 Pro (FMUv4pro) · PX4 User Guide

mrpollo, 2020-08-06:
Link to pixhawk.drotek.com is no longer valid.

[image]

Bug Page: [Drotek Pixhawk 3 Pro (FMUv4pro) · PX4 User Guide](http://docs.px4.io/master/en/flight_controller/pixhawk3_pro.html)

hamishwillee, 2020-08-10:
Hi @Paul-DROTEK The link to "Main hardware documentation" formerly at https://pixhawk.drotek.com/en/ no longer works. Can you point me to the updated link? Obviously I don't know exactly what is at that link, but I am expecting design files like at https://github.com/pixhawk/Hardware

Paul-DROTEK, 2020-08-10:
Hi @mrpollo and thanks @hamishwillee for reaching out to me.

I am not really sure what happen with the dead link above.
Nonetheless you are still able to reach our gitbook product page related to the Pixhawk 3 Pro by using the link below :
https://drotek.gitbook.io/pixhawk-3-pro/

The hardware and pinout ressources are accessible here : https://drotek.gitbook.io/pixhawk-3-pro/hardware

Please @hamishwillee let me know wether the links provided are okay. @mrpollo feel free to reach us out [here ](https://electronics.drotek.com/after-sale-service/) if you need any further details.

hamishwillee, 2020-08-11:
Thanks @Paul-DROTEK - those links look good. 

FWIW They probably actually wanted open-design schematics, but I am pretty sure that these were not supplied (ever) for this FMU version/board. Your links are the "right replacements" that I wanted here :-)

PR #785 (2020-08-11): Fix link to drotek docs on Pixhawk3pro
Fixes #784
closed 2020-08-11 by hamishwillee