PX4/PX4-user_guide #869: Links to older versions broken with new build

RicardoM17, 2020-11-09:
The links from current master no longer redirect to the same page in the older version but instead redirect to the main docs page.

Also for V1.10 it redirects to the QGC docs instead.

Issue raised by Andrew Saba on PX4 slack #documentation channel

> FYI, the current master documentation page, when you switch versions it redirects back to the home page of the documentation. Older versions would keep you on the same doc page but grab an older version. Is this intentional/something thats not easy to fix?
> Also, more of a bug, when switching to the v1.10 version, it redirects to the QGC docs page. Im guessing thats a bug? I can open an issue if you want. Tested in Firefox on ubuntu 18.04. All other versions go to PX4 docs as expected.

@hamishwillee FYI

hamishwillee, 2020-11-09:
> Older versions would keep you on the same doc page but grab an older version. Is this intentional/something thats not easy to fix?

Known issue. Not easy to fix.

> When switching to the v1.10 version, it redirects to the QGC docs page. Im guessing thats a bug? I can open an issue if you want. Tested in Firefox on ubuntu 18.04. All other versions go to PX4 docs as expected.

Yes, that's a bug. I will fix tomorrow.

hamishwillee, 2020-11-10:
OK, so the qgc link is fixed by https://github.com/PX4/PX4-user_guide/commit/45a1c39f9c4582ac69eb58db709a2987b0dee1d6

hamishwillee, 2020-11-10:
I am closing this. The requirement for the version switcher is more precisely noted in https://github.com/PX4/PX4-user_guide/issues/872
Thanks!

closed 2020-11-10 by hamishwillee