PX4/PX4-user_guide #928: Anchor link does not expand side menu

elpiel, 2020-11-30:
When opening a link with anchor, e.g.
https://docs.px4.io/master/en/simulation/#sitl-simulation-environment

The side menu doesn't expand on the `Development -> Simulation`

RicardoM17, 2020-11-30:
The same happens here. Thanks for the feedback @elpiel . We recently changed our documentation build tools so it's "normal" that we have these issues. So it's good when people point them up.

elpiel, 2020-11-30:
No problems, looks nice and it's understandable to have some wrinkles to iron ;)

hamishwillee, 2020-12-02:
Well, this is a weird one. The vast majority of cases do work:
- https://docs.px4.io/master/en/modules/modules_command.html#dyn
- https://docs.px4.io/master/en/concept/flight_modes.html#flight-mode-summary
- https://docs.px4.io/master/en/flight_stack/controller_diagrams.html#multicopter-control-architecture
- https://docs.px4.io/master/en/debug/consoles.html#system-console-vs-shells

But then some don't:
- https://docs.px4.io/master/en/debug/consoles.html#system-console-vs-shells
- https://docs.px4.io/master/en/concept/architecture.html#high-level-software-architecture
 - https://docs.px4.io/master/en/modules/modules_main.html#categories

And for the ones that [don't](https://docs.px4.io/master/en/debug/consoles.html#system-console-vs-shells) - going [deeper](https://docs.px4.io/master/en/debug/mavlink_shell.html#mavlink-shell-py) will work.

I'm unable to work out what it is that makes some work and some not.

hamishwillee, 2025-01-29:
It works now on current infrastructure - e.g. try https://docs.px4.io/main/en/simulation/#sitl-simulation-environment

closed 2025-01-29 by hamishwillee