PX4/PX4-user_guide #941: Doc Bug: Acro (MC) · PX4 v1.8.2 User Guide

whataberk, 2020-12-02:
In the expo-describing equation: y=r(x^3+x(1−f))(1−g)/(1−g|x|) it does not appear to match the description or the code. I believe that it should be y=r(**f***x^3+x(1−f))(1−g)/(1−g|x|) based on the code from [here](https://github.com/PX4/PX4-Autopilot/blob/378c0b509398005cc38c4e85caf3050dcd6b9a80/src/lib/mathlib/math/Functions.hpp#L65), copied below:
```cpp
template<typename T>
const T expo(const T &value, const T &e)
{
	T x = constrain(value, (T) - 1, (T) 1);
	T ec = constrain(e, (T) 0, (T) 1);
	return (1 - ec) * x + ec * x * x * x;
}
```

This appears to affect all extant versions of the guide (1.8-1.11).

Bug Page: [Acro (MC) · PX4 v1.8.2 User Guide](https://docs.px4.io/v1.8.2/en/flight_modes/acro_mc.html)

RicardoM17, 2020-12-08:
Hi @whataberk thanks for raising this. I see that this was merged to master - V1.10 in #942 .

@hamishwillee I closed this. Is that ok or are you planning to back port to even older versions? At this point it doesn't make much sense to support such old firmware/documentation IMO.

closed 2020-12-08 by RicardoM17

hamishwillee, 2020-12-08:
@RicardoM17 Thanks, I merged the fix and backported to the last couple of releases. Just missed that this was linked :-)