PX4/PX4-user_guide #987: Durandal Pinmap PDF currently linked, possibly wrong

rotorman, 2021-01-02:
Please check the [Stm32H743IIK_pinmap.pdf](https://github.com/PX4/px4_user_guide/raw/master/assets/flight_controller/durandal/Stm32H743IIK_pinmap.pdf) (Durandal Pinmap) linked PDF in [https://docs.px4.io/master/en/assembly/quick_start_durandal.html#further-information](https://docs.px4.io/master/en/assembly/quick_start_durandal.html#further-information).

The PDF that is currently linked, reads in the header line of each page "STM32F765IIK" and lists 140 pins, whereas Durandal flight controller is using a more newer H743 microcontroller in UFBGA176+25 (10x10 mm) package (more pins than 140).

Either the PDF is for another flight controller (which I assume), or there is a typo in the PDF.
In either case, I would appreciate if this got cleared/fixed - thank you!

hamishwillee, 2021-01-20:
@vincentpoont2 The file referred to here is the one hosted at http://www.holybro.com/manual/Durandal-Pinouts.pdf

Can you please comment?

rotorman, 2021-01-20:
I am not referring to this file, but the following:
[https://raw.githubusercontent.com/PX4/PX4-user_guide/master/assets/flight_controller/durandal/Stm32H743IIK_pinmap.pdf](https://raw.githubusercontent.com/PX4/PX4-user_guide/master/assets/flight_controller/durandal/Stm32H743IIK_pinmap.pdf)

vincentpoont2, 2021-01-20:
@rotorman 
The header of the pinout pdf is incorrect, but the pin information should be correct for Durandal.

I will reupload with the correct header.

vincentpoont2, 2021-01-20:
[STM32H743IIK_pinout.pdf](https://github.com/PX4/PX4-user_guide/files/5841103/STM32H743IIK_pinout.pdf)
@hamishwillee can you update the pdf with this?

rotorman, 2021-01-20:
I compared the document with ArduPilot hardware definitions here:
[https://raw.githubusercontent.com/ArduPilot/ardupilot/master/libraries/AP_HAL_ChibiOS/hwdef/Durandal/hwdef.dat](https://raw.githubusercontent.com/ArduPilot/ardupilot/master/libraries/AP_HAL_ChibiOS/hwdef/Durandal/hwdef.dat)
and saw some minor differences, here listed all what I saw, either missing in PDF or missing in ArduPilot hwdef.dat or a conflict between them:

PB10 nSPI5_RESET_EXTERNAL1 OUTPUT HIGH
PC5 DRDY4_ICM20602 INPUT
PE4  VDD_3V3_SPEKTRUM_POWER_EN
PE5 TIM9_CH1 BUZZER_1 vs. TIM15_CH1 TIM15 GPIO(77)
PE10 GND_VIA_10K
PF3 ICM20602_CS CS SPEED_VERYLOW ALARM
PG0 HW_VER_DRIVE (PDF lists PH14)
PG3 NPOWER_IN_C
PG5 VDD_5V_RC_EN OUTPUT HIGH SPEED_LOW LOW GPIO(70)
PG6 VDD_5V_WIFI_EN OUTPUT HIGH SPEED_LOW LOW GPIO(71)
PH4 GPIO_CAN3_SILENT OUTPUT PUSHPULL
PH14 HW_REV_DRIVE (AP lists PG0)
PI0 FMU_SPARE_4 INPUT GPIO(61)
PI5 PL5 (<- typo PL instead of PI) GND_VIA_10K

It would be great if you could check them at the same time and either fix in PDF, or ArduPilot hwdef.dat or both if necessary. Thank you!

hamishwillee, 2021-01-24:
@vincentpoont2 Yes, on Wednesday :-). Do you want to check out the possible errors highlighted by @RicardoM17 in https://github.com/PX4/PX4-user_guide/issues/987#issuecomment-763454660 first?

RicardoM17, 2021-01-25:
> @vincentpoont2 Yes, on Wednesday :-). Do you want to check out the possible errors highlighted by @RicardoM17 in [#987 (comment)](https://github.com/PX4/PX4-user_guide/issues/987#issuecomment-763454660) first?

wrong tag @hamishwillee. I believe you mean @rotorman instead :sweat_smile:

hamishwillee, 2021-01-26:
@vincentpoont2 To be very precise, can you please confirm:
1. you want me to replace all links to http://www.holybro.com/manual/Durandal-Pinouts.pdf to STM32H743IIK_pinout.pdf (linked in https://github.com/PX4/PX4-user_guide/issues/987#issuecomment-763432819). (note, I have put this into source tree [here](https://github.com/PX4/PX4-user_guide/raw/master/assets/flight_controller/durandal/STM32H743IIK_pinout.pdf).
2. Can you comment on https://github.com/PX4/PX4-user_guide/issues/987#issuecomment-763454660 ?

vincentpoont2, 2021-01-27:
[image]
This document needs to be removed and replaced with this: 
[STM32H743IIK_pinmap.pdf](https://github.com/PX4/PX4-user_guide/files/5877343/STM32H743IIK_pinmap.pdf)

Thanks!

@rotorman I have confirmed with the Holybro engineer that one attached here is correct, it was just an typo on the header. He will look at the conflict/mistake on the Ardupilot side soon.

hamishwillee, 2021-01-27:
Thanks @vincentpoont2 Done in https://github.com/PX4/PX4-user_guide/commit/8d30450b9383085190160bf365400555e8fbcfc8 and https://github.com/PX4/PX4-user_guide/commit/bdf0111e14bf85fa1ea578b09cb5eac36817a5e1 (I missed the name change)

commit 8d30450b (2021-01-27): Fix Durandal further info bitmap

Fixes #987
closed 2021-01-27 by hamishwillee