PX4/PX4-user_guide #1000: Wrong port PX4 SITL - QGC

boris-gu, 2021-01-19:
The documentation says that for SITL it is port 14570, although in fact it is port 18570.
https://docs.px4.io/master/en/test_and_ci/docker.html#qgroundcontrol
The same error is present in the image here:
https://docs.px4.io/master/en/simulation/ros_interface.html#ros-with-gazebo-simulation

[image]

boris-gu, 2021-01-19:
As a result, I managed to connect QGC to the simulation in the docker container by writing these values:
[image]

hamishwillee, 2021-01-20:
The SITL ports are [documented here](https://docs.px4.io/master/en/simulation/#sitl-simulation-environment) - as these indicate QGC normally connects on 14550

Can you please try that? Either way the 14570 is probably wrong.

boris-gu, 2021-01-20:
@hamishwillee Forgive me, I do not quite understand this phrase: "18570 is supposed to be for sending to rather than listening for connections". English is not my native language.
If I understand correctly, it works like this - QGC listens on port 14550, on which it receives messages from the PX4 autopilot. In turn, the PX4 autopilot listens on port 18570 to receive information from the QGC.

hamishwillee, 2021-01-20:
@boris-gu My understanding is that what normally happens is that QGC "listens" to port 14550 on the computer running PX4 - ie it waits for mavlink traffic before it starts broadcasting its heartbeat. The address it sends the UDP traffic back to is in the packet it gets which I presume is 18570. 

I could be wrong on this, but I do know that 14550 works in almost every scenario. So I am hoping you can try that out and confirm.

 I have also asked for clarification in the [#simulation slack channel](https://px4.slack.com/archives/CL3LV4FUH/p1611108528005100))

hamishwillee, 2021-01-24:
@bkueng can you advise on this one? 

CI/Docker docs suggest connecting to PX4 hosted on docker using 14570. The Simulation docs suggest 14550. This issue suggests 18570, and the CI docs do mention configuration file in which this port does appear.

hamishwillee, 2021-02-10:
@julianoes Can you comment on this. Getting a similar comment in #1000

PR #1047 (2021-02-18): Fix up simulator port docs and diagram
Port used for GCS local port is wrong. Simulation overview missing comments around how multiple instances work, and also using RTPS/DDS instead of MAVLink simulation API. 

- Fixes #1035
- Fixes #1000

This is currently in draft pending response to https://github.com/PX4/PX4-user_guide/issues/1035#issuecomment-777095536.
closed 2021-02-18 by hamishwillee