PX4/PX4-user_guide #1025: Flight modes supported by Boats and Rovers

hamishwillee, 2021-01-31:
Docs sayy nothing about even what modes are supported on boats and rovers - e.g. in
- https://docs.px4.io/master/en/getting_started/flight_modes.html
- https://docs.px4.io/master/en/flight_modes/ (and sub-topics)

General questions we'd like to be able to answer for any vehicle
- what modes are supported by rover and boat? What modes from the set in MC are not?
- Where modes are supported, how are they different than for MC? For example on position mode is a boat like FW - ie loiters at point, or does it just wait until it drifts outside a radius then go to middle, or something else?
- What are the default stick mappings

@Jaeyoung-Lim Do you know the answers, or someone we could talk to?

Jaeyoung-Lim, 2021-02-11:
@hamishwillee The answer is actually just manual mode and mission mode.

I am currently adding a attitude / bodyrate controller which will add support for acro mode / stabilized mode

hamishwillee, 2025-01-29:
Closing. This is being dealt with for Rovers, and when boat is properly supported we'll look at this then.

closed 2025-01-29 by hamishwillee