PX4/PX4-user_guide #1026: Offboard Mode, PX4, rospy, plane/rover examples?

RotorBoy2020, 2021-02-01:
Hello,

Are there any python examples of the offboard control for planes and rovers? I have written my own code based [off this tutorial](https://docs.px4.io/master/en/ros/mavros_offboard.html) for MAVROS quadcopter control, and would be interested in some further examples for non-quad frames. 

I was able to follow this [link which shows which messages are possible for each frame.](http://docs.px4.io/master/en/flight_modes/offboard.html) However, I am unclear as to how to send these commands via MAVROS, as these are clearly MAVLINK messages. 

The message that seems of interest to me is: https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED. But how can I call that from ROS using MAVROS?

Any help would be greatly appreciated!

hamishwillee, 2021-02-02:
@RotorBoy2020 The docs have [MAVROS Offboard control example](https://docs.px4.io/master/en/ros/mavros_offboard.html) which is a C example. That points to similar examples in Python here: https://github.com/PX4/PX4-Autopilot/tree/master/integrationtests/python_src/px4_it/mavros

While these are for copter, the way that you integrate with the MAVLink position target messages should be exactly the same from a ROS point of view. The actual values you send in the messages depends on the links supported by each frame (that you point to above).

There are no other examples that I know about. @Jaeyoung-Lim or @TSC21 may know of others.

If you do end up generating an example, then please add it :-). I'm closing this now, but you might also consider posting in the mavros discussion boards.

closed 2021-02-02 by hamishwillee

RotorBoy2020, 2021-02-03:
Ok thank you, I'll post my question on the MAVROS discussion board 

I generated the messages the same way that I did for the copter but the plane doesn't fly as expected, so something is amiss