PX4/PX4-user_guide #1030: Visual Studio Code IDE

mengchaoheng, 2021-02-03:
In Visual Studio Code IDE, I don’t see the jlink (px4_fmu-v5) option. How do I add a debug configuration?

hamishwillee, 2021-02-04:
@mengchaoheng I think it just appears when you have the Firmware open and are connected via JLink (as per http://docs.px4.io/master/en/dev_setup/vscode.html#hardware-debugging).

@dagar can you confirm?

mengchaoheng, 2021-02-04:
> @mengchaoheng I think it just appears when you have the Firmware open and are connected via JLink (as per http://docs.px4.io/master/en/dev_setup/vscode.html#hardware-debugging).
> 
> @dagar can you confirm?

In the commit 850d0bc588af79186286652af4c8293daafd2c4c version, after executing make px4_fmu-v5 in the terminal, the launch.json file is not generated in the vscode folder. In the recent commit 3ab4213314c95d24f902dae92535fecd8c629f25, the launch.json file is automatically generated. This version allows me to easily use jlink for hardware debugging in vscode. Now the problem is that I have done a lot of work in commit 850d0bc588af79186286652af4c8293daafd2c4c version. I don't know how to transfer these work to the latest version. I guess I need to learn git. Do you have any suggestions? Or how can I fix this bug in commit 850d0bc588af79186286652af4c8293daafd2c4c version?

hamishwillee, 2021-02-04:
@mengchaoheng Can you please raise on the discussion boards or slack. I appreciate this is a docs issue too, but you will likely get faster response there.

dagar, 2021-02-05:
@mengchaoheng 

This is where it's generated in current master.

https://github.com/PX4/PX4-Autopilot/blob/a399bb9894b1d93639b61cf59866695c55806153/platforms/nuttx/CMakeLists.txt#L212-L288

hamishwillee, 2021-02-07:
From Dagar:

> it's the cmake configuration that generates the vscode launch.json. I.e when you select the config/variant (eg [cmake] px4_fmu-v5_default this will generate the right option.

hamishwillee, 2021-02-10:
Thanks @mengchaoheng for reporting. Fixed in #1046 - should appear in the built version later today.

PR #1046 (2021-02-10): vscode: clarify when you see the jlink option
Fixes #1030.

FYI @dagar This says you need to select a firmware cmake target to get jlink debug options (which is what you told me). I'm self merging, but would appreciate a sanity check.
closed 2021-02-10 by hamishwillee