PX4/PX4-user_guide #1035: Inaccurate SITL Port Number

aseyfi, 2021-02-05:
Hello,

I was looking at this [SITL block diagram](https://docs.px4.io/master/en/simulation/#sitl-simulation-environment) and it seems PX4 is using ports 14570 and 14580 for its communications, however, when I run PX4 in SITL mode, I see these messages:
```
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030 
```
Doesn't it mean PX4 uses 18570 (not 14570) to connect to 14550?

Thank you,
Ahmad

hamishwillee, 2021-02-10:
My understand of UDP is that there is a send (remote) port and a listen port. 
- PX4 sends traffic for the ground station to port 14550 - so this is where QGC listens for traffic.
- PX4 sends traffic to 14540 for offboard APIs like the MAVSDK or ROS - so this is where MAVSDK would listen for traffic.

In theory system might also start a connection by sending traffic to PX4's receive port, but these ports, while labelled on the diagram are not usually used, and are not guaranteed not to change. 

The port 18570 is the recieve port that PX4 is listening for. I think that means that the diagram is mislabeled as shown. @julianoes could you confirm?

[image]

julianoes, 2021-02-10:
Correct, this has changed. @hamishwillee and you had even reminded that this would require a change in the docs :smirk:.

https://github.com/PX4/PX4-Autopilot/commit/c89766c8aba51212133336499f4c5ee8144b1a6f
https://github.com/PX4/PX4-Autopilot/pull/14178#issuecomment-594340748

hamishwillee, 2021-02-10:
@julianoes Long time ago, and I still don't _fully_ get it. @Jaeyoung-Lim , perhaps you can also comment.

So my understanding is (as above). 
- PX4 defines local and remote UDP ports. 
- A local port is essentially what PX4 listens for traffic on. A remote port is where it broadcasts to. 
- A GCS listens for traffic on the remote port (e.g. HEARTBEAT). The UDP messages it gets include the reply address, and that will be the "local" address to PX4.
- The code seems to indicate that:
  - Ground stations such as *QGroundControl* connect to all instances using the normal UDP port 14550 (all traffic goes to the same port). The local port iterates up from 18570.
  - Developer APIs such as *MAVSDK* or *MAVROS* connect on the UDP port 14540 (first instance), UDP port 14541 (second instance), to 14549, which is then used for *every other instance*.  The local port iterates up from 14580.

Questions:
- is the above all correct?
- Why can all ground stations connect to the same remote port for all instances, but APIs need to iterate the remote port? Further, at instance 14549 APIs stop iterating the port ... which is confusing.
- Why do we need multiple local ports rather than just one?
- My understanding is that all the differentiation can be done on MAVLink Sys ID, which is why the previous 2 questions.

julianoes, 2021-02-11:
@hamishwillee 

> So my understanding is (as above).

Yes, that's all correct except I would phrase this differentely:

> A GCS listens for traffic on the remote port (e.g. HEARTBEAT).

A GCS listens for traffic on PX4's remote port (which is the GCS' local port)

> Why can all ground stations connect to the same remote port for all instances, but APIs need to iterate the remote port? Further, at instance 14549 APIs stop iterating the port ... which is confusing.

I don't think there is a reason. It's just how it came to be. So, we could make it consistent, or we could leave it as is not to break anyone.

> Why do we need multiple local ports rather than just one?

Multiple processes can send messages to the same remote port but they can't bind to the same local port. If you try to bind to a local port that someone else is already bound to you get an error (see EINVAL in https://pubs.opengroup.org/onlinepubs/009695399/functions/bind.html).

So to say it differently we decided to use 14550 as a default convenience port but for discovery but we can't do that the other way round.

Often, in these setups implementations no local port is specified. This means that the OS assigns you some random port which avoids conflicts. So, my preferred solution would be to remove all these local ports which would prevent having to shift them around and keep the docs in sync, etc..
However, there are some cases where we rely on knowing the local port:
1. As far as I know the port number can passed to MAVROS to initiate the connection. So basically, MAVROS will start sending heartbeats to that port and actively initiate the connection, instead of just waiting on 14550 or 14540 for an autopilot to appear.
2. For some cases knowing the port can help to break through barriers like networking into a docker container or virtualbox. This is a bit hand-wavy but I believe it can help.

The current state is like that because mavlink does not specify discovery but historically a ground station can expect an autopilot to connect on 14550.

For intra-network discovery we could think of adding a discovery port that can be used in both/all direction to find new participants. Once two components have discovered each other, they can then do a handshake to agree on another port pair and move there, leaving the discovery port open for more discoveries. I think that's high level what is described in http://hintjens.com/blog:32.

For discovery outside of the network, you would probably need something like IP and/or DNS, I'm not too familiar with all of that.

So, having said all that, I suggest we change the docs to match what's there now but we don't change anything around the 14550 and 14540 inconsistency for now but consider picking up the discovery topic.

PR #1047 (2021-02-18): Fix up simulator port docs and diagram
Port used for GCS local port is wrong. Simulation overview missing comments around how multiple instances work, and also using RTPS/DDS instead of MAVLink simulation API. 

- Fixes #1035
- Fixes #1000

This is currently in draft pending response to https://github.com/PX4/PX4-user_guide/issues/1035#issuecomment-777095536.
closed 2021-02-18 by hamishwillee