PX4/PX4-user_guide #1055: still have error s while building PX4 after clean install UBUNTU

helloE9, 2021-02-13:
c/modules/battery_status/module.yaml /home/kpros/PX4-Autopilot/src/modules/mavlink/module.yaml
Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

ninja: build stopped: subcommand failed.
make: *** [Makefile:219: px4_fmu-v3_default] Error 1

------------------------
so, I did 

sudo -H pip3 install --user pyserial empy toml numpy pandas jinja2 pyyaml pyros-genmsg packaging

Requirement already satisfied: pyserial in /root/.local/lib/python3.8/site-packages (3.5)
Requirement already satisfied: empy in /root/.local/lib/python3.8/site-packages (3.3.4)
Requirement already satisfied: toml in /root/.local/lib/python3.8/site-packages (0.10.2)
Requirement already satisfied: numpy in /root/.local/lib/python3.8/site-packages (1.20.1)
Requirement already satisfied: pandas in /root/.local/lib/python3.8/site-packages (1.2.2)
Requirement already satisfied: jinja2 in /root/.local/lib/python3.8/site-packages (2.11.3)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (5.3.1)
Requirement already satisfied: pyros-genmsg in /root/.local/lib/python3.8/site-packages (0.5.8)
Requirement already satisfied: packaging in /root/.local/lib/python3.8/site-packages (20.9)
Requirement already satisfied: python-dateutil>=2.7.3 in /usr/lib/python3/dist-packages (from pandas) (2.7.3)
Requirement already satisfied: pytz>=2017.3 in /usr/lib/python3/dist-packages (from pandas) (2019.3)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/lib/python3/dist-packages (from jinja2) (1.1.0)
Requirement already satisfied: pyparsing>=2.0.2 in /root/.local/lib/python3.8/site-packages (from packaging) (2.4.7)

//////////////////
but I got the same error 

Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

[10/1407] Generating git version header
ninja: build stopped: subcommand failed.
make: *** [Makefile:219: px4_fmu-v3_default] Error 1

Jaeyoung-Lim, 2021-02-13:
```
Tools/ubuntu/setup.sh
```

helloE9, 2021-02-13:
> ```
> Tools/ubuntu/setup.sh
> ```

yes. of course I did it to setup environment for build and checked version of gcc also.

Jaeyoung-Lim, 2021-02-13:
Then you need to check why you cant find the python packages which are installed using the script

hamishwillee, 2021-02-17:
In the past I have seen these kinds of errors on systems that have python2 already installed. If you really started on clean Ubuntu 20.04 then this should work (i just tried it on a fresh install in a VM). 

1. Please confirm your ubuntu version
2. Also that you cloned the firmware then from root ran `./Tools/ubunu.sh`. 

This must be very irritating for you sorry! Not sure how to help you debug. We wrote this script (and I believe run it in CI) so we don't have to debug this sort of thing.

helloE9, 2021-02-20:
> In the past I have seen these kinds of errors on systems that have python2 already installed. If you really started on clean Ubuntu 20.04 then this should work (i just tried it on a fresh install in [...]

I've installed Ubuntu 20.04 again and ran the 'ubuntu.sh' 
The errors all cleared. 
There was some mistakes during installation of other python projects.

Thanks for your help!!

helloE9, 2021-02-20:
> Then you need to check why you cant find the python packages which are installed using the script

Thank you for your help.
It has been cleared.
I tried to uninstall and re-install old python but it's been failed.
so, I've install ubuntu again and all cleared. :)

closed 2021-02-20 by helloE9