PX4/PX4-user_guide #1091: Issue with gimbal control on PX4

cujo612, 2021-03-04:
Hello,

I am working with a CUAV V5+, an SBGC v3.6 "Extended" gimbal controller, and some gimbal motors. I have been able to control the gimbal motors properly through the simpleBGC app but not through QGroundControl. I am trying to use mavlink commands to move the gimbal in QGC. 

I tried the  MAV_CMD_DO_GIMBAL_MANAGER_CONFIGURE command in the mavlink console but it said command not found

I have the following settings configured:

In GCS:
MNT_MODE_IN: MAVLINK_DO_MOUNT
MNT_MODE_OUT: MAVLINK
MAV_1_CONFIG: TELEM2
MAV_1_MODE: NORMAL
MAV_1_RATE: 11520 B/s

In SimpleBGC:
RC Settings
RC_ROLL pin mode: Serial port (Serial API, etc)
ROLL/PITCH/YAW: no input

External IMU
Model: MavLink FC channel 1
Connection: UART1
Serial port: UART1
System ID/Component Id: 1 154
Port settings: 115200, none parity
send heartbeat: yes
Query RC data: yes

hamishwillee, 2021-03-07:
Best to open these queries on Slack or discussion board. Also what version of PX4 you working on?

@julianoes Can you confirm what versions of PX4 the new gimbal manager should work? Just very latest master?

julianoes, 2021-03-08:
@cujo612 as @hamishwillee suggested, it would probably be best to please ask in Slack in the #gimbal channel and we can try to talk this through there.

The questions I have are if your gimbal controller already supports the MAVLink gimbal v2 protocol, and what PX4 version you are using.

cujo612, 2021-03-08:
Thanks @julianoes and @hamishwillee for the responses. It looks like I need an invite to join the slack. 

The firmware is 1.11.0dev and the gimbal controller is up to date on software and should support MAVLink v2 protocal

julianoes, 2021-03-08:
slack.px4.io to get an invite.

MAVLink v2 protocal is not the same as MAVLink gimbal protocol v2. I suspect it is still using v1.

Also see:
https://mavlink.io/en/services/gimbal_v2.html
https://mavlink.io/en/services/gimbal.html

cujo612, 2022-03-09:
@evanatherton1 Unfortunately I did not

hamishwillee, 2022-03-16:
Closing. Please post in slack or the discussion boards. We're more than happy to help you there.

closed 2022-03-16 by hamishwillee