PX4/PX4-user_guide #1133: v1.12 Updates: More intuitive stick feel in Position mode

mrpollo, 2021-03-22:
We need to update the documentation for position control with the latest changes that are coming to v1.12, 

**More intuitive stick feel in Position mode**
* Horizontal stick input mapped to acceleration instead of velocity setpoints
* Removes unexpected tilt changes upon reaching travel speed velocity
* Intuitive shunting e.g. when landing

> From @hamishwillee This needs to go into docs in Position Mode here: http://docs.px4.io/master/en/flight_modes/position_mc.html and possibly also here: http://docs.px4.io/master/en/getting_started/flight_modes.html#position-mode-mc.

https://github.com/PX4/PX4-user_guide/pull/1122/files/38b96aa2c14f34cf4648d2824a79f39d76b43d0a#r596419982

hamishwillee, 2021-03-24:
There are two aspect to this; explaining how it is different to before (and why it is more natural), and updating docs to show how it works now.

- [ ] Horizontal stick input mapped to acceleration instead of velocity setpoints

[image]

So I think that for yaw (spin) and left/right:
- previously you'd move at increasing speed from 0 to max, then let go to stop
- now you accelerate at increasing rate. Let go to return go to position control and presumably stop very quicly 

I don't see why the new behaviour is more intuitive other than it is faster and probably gets you where you want to go faster and with much smaller adjustment. Can you explain why it is?

With respect to the docs change though, probably need to change these two points:
FROM
- Roll/Pitch sticks control speed over ground in left-right and forward-back directions (respectively) relative to the "front" of the vehicle.
- Yaw stick controls rate of angular rotation above the horizontal plane.

TO 
- Pitch stick controls speed over ground forward-back direction (relative to the "front" of the vehicle).
- Roll stick controls the acceleration over ground in left-right direction (relative to the "front" of the vehicle).
- Yaw stick controls the rate of change of angular rotation above the horizontal plane (relative to the "front" of the vehicle).

Right?

- [ ] Removes unexpected tilt changes upon reaching travel speed velocity

Mostly this is "just for my information". But what unexpected tilt changes/why/how/how does this help? 
I know the vehicle tilts when it moves, and I believe that tilt is related to speed. So I am assuming that what happened previously was that if you push stick to a position, the vehicle accelerates to that speed then overshoots the speed and wobbles. 
Not sure what happens now - presumably there can't be an overshoot somehow?

- [ ] Intuitive shunting e.g. when landing

I have never heard this term. What does it mean? How has it changed since previously? Does it only affect landing?

MaEtUgR, 2021-03-29:
@hamishwillee Thanks for your valuable feedback. This gives me a new perspective 👍🏻
- Yaw and Altitude (left stick on mode 2) stayed unchanged. Left, right, forward, backwards or in other word horizontal movement is what the comments are about.
- Tilt is more or less directly related to acceleration. Before if the stick gave 5 m/s it would tilt hard until it reached exactly 5 m/s and then twitch back to cruise at that speed. No overshoot no wobble but still not what feels intuitive. Compare to the gas pedal setting cruise control for your car. You'll still reach the same maximum speed, maybe even a bit slower but the reactions to your stick make more sense because it's more like tossing the drone around.

Does that help at all? I 'd be happy to explain further, maybe some simple graphs help.

hamishwillee, 2021-03-29:
Thanks @MaEtUgR , I'll think about this tomorrow. But my first reaction is that this is helpful. 

I think what you're saying is that the tilt of the vehicle now corresponds to the stick position, and that feels more natural. The comparison to a car might be that previously we have a kind of bang-bang where we always accelerate at maximum rate. Now we accelerate at a constant user-set rate - the whole thing will feel less "jerky". Is that about right?

hamishwillee, 2021-03-30:
OK, I think I get this. The changes for docs are relatively minimal and done in #1158.

Open question is what you mean by "Intuitive shunting e.g. when landing"

I think the release notes are basically OK other than that in the current version.

PR #1158 (2021-04-21): Position mode stick
Fixes #1133

@MaEtUgR Because the docs are versioned, we document the current behaviour rather than the "change". Essentially this means changing the text where we say that we control movement over ground from speed to acceleration. This happens in three places:
- [Getting Started > Flight Modes](https://docs.px4.io/master/en/getting_started/flight_modes.html#position-mode-mc)
- [Flight Modes Overview page](https://docs.px4.io/master/en/flight_modes/)
  - [Position Mode (MC)](https://docs.px4.io/master/en/flight_modes/position_mc.html)

Mostly pretty straightforward, but so [... cut]
closed 2021-04-21 by hamishwillee