PX4/PX4-user_guide #1135: v1.12 Updates: Multi-EKF enabled by default

mrpollo, 2021-03-22:
We need to document this feature on our user guide, although this feature has been part of previous releases, it was turned off by default, and was never properly documented

https://github.com/PX4/PX4-Autopilot/pull/14650

https://github.com/PX4/PX4-Autopilot/pull/16079

hamishwillee, 2021-03-24:
FYI the docs we currently have about EKF are here: http://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html (there are a bunch more minor docs that mention it, - eg. system replay - but nothing useful. 

I'd say we minimally need two things.
1. That doc updated, if anything changed w.r.t tuning. That might affect lots of things - e.g. things like optical flow are impacted by EKF tuning for their position.
2. Any other info needed for an END user.
3. Developer docs, explaining the architecture - probably in http://docs.px4.io/master/en/concept/architecture.html . Anything else a developer might need to do to maintain, debug, or interconnect with the EKF from other code 

All of this has to come from someone with expertise. Yes, @dagar, I do mean you :-)

Minimally it would be good to know if there are any impacts for end users in terms of tuning.

hamishwillee, 2021-04-28:
@mrpollo Great news. Paul updated these docs - you can link to them here: http://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#running-a-single-ekf-instance  (actually it is the following heading on the same page, but I selected this because it makes it clear that SINGLE EKF is still the default.

There is also update to the stuff related to default parameters in http://docs.px4.io/master/en/dev_airframes/adding_a_new_frame.html#config-file - basically use `param set-default` rather than `param set` (that is still used, but only for updates - so clear differentiation of setting the default value).

I assume you can update the release notes right?

hamishwillee, 2021-05-06:
@mrpollo Closing this as work is done. But I am assuming you are sorting out any cross linking from the release notes.

closed 2021-05-06 by hamishwillee