PX4/PX4-user_guide #1137: v1.12 Updates: Commander Separate out manual control setpoint processing

mrpollo, 2021-03-22:
We need to update the documentation to include these changes

https://github.com/PX4/PX4-Autopilot/pull/16878

hamishwillee, 2021-03-24:
Hi @MaEtUgR 

Looks like changes are:
- [ ] RC override: option to ignore throttle stick 
    - ie `COM_RC_OVERRIDE` now has new option to turn off the RC transition to position mode from manual mode.
    - UPdate required for param info in all the auto modes: https://docs.px4.io/master/en/flight_modes/mission.html#mission-parameters , https://docs.px4.io/master/en/flight_modes/takeoff.html#multi-copter-mc , https://docs.px4.io/master/en/flight_modes/land.html#multi-copter-mc , https://docs.px4.io/master/en/flight_modes/return.html#parameters , https://docs.px4.io/master/en/flight_modes/hold.html#multi-copter-mc
    - Matthias, my assumption is that this does not affect orbit mode, follow me, or any of the FW modes. Correct?
- [ ] RC override: based on change (not centered stick)
  - So this is related to `COM_RC_STICK_OV`, the the distance the sticks need to move to trigger the RC override in the point above.
  - Matthias, is there a "time" within which the movement has to happen?
  - Matthias. Also is it distance from first point, or total  (ie if I make 200 very small moves back and forth within time that equals this, will that count? Or is it just, detect movement and record start point, if we exceed this distance away within time we trigger?)
  - I think we'd just add the param to the same place as the first point.

Not sure what these mean
- [ ] RC arming: based on Hysteresis instead of "loop ticker"
- [ ] RC arming: simplify the logic
- manual_control_setpoint message

... But I see a change to `COM_ARM_SWISBTN` . The arming docs are here: https://docs.px4.io/master/en/advanced_config/prearm_arm_disarm.html#arming-button-switch
They look right, but maybe just a tidy up to match the changes to `COM_ARM_SWISBTN`?

What else am I missing? Any further suggestions?

MaEtUgR, 2021-03-29:
@hamishwillee 
- COM_RC_OVERRIDE has the option to disable for offbaord or auto modes independently and also to ignore the throttle stick axis. We should add that (if it wasn't already in 1.11).
- Stick trigger based on not being center was also the way we released 1.11 so for the stable release user there's no change.
- The timing and separation comments are pure refactoring. There's no influence on the pilot (yet) but the developer can change it more easy.

hamishwillee, 2021-03-31:
Thanks @MaEtUgR 

Thanks very much. So in summary, yes, all of this was documented in v1.11. I did some tidy up in https://github.com/PX4/PX4-user_guide/pull/1159 because that wasn't immediately obvious to me.

I'm going to close this because my understanding from above is that there is no docs impact that has not already been captured.

closed 2021-03-31 by hamishwillee