PX4/PX4-user_guide #1177: Error on advanced parameters page

TaeronSystems, 2021-04-06:
On the advanced parameters page, the COM_ARM_WO_GPS value definitions are reversed. The values are correct on release "v1.11" but wrong on "Master".

The problem occurred during the commit:
["Update PX4 Firmware metadata Tue Feb 23 00:43:10 UTC 2021"](https://github.com/PX4/PX4-user_guide/blob/4dd371577e772e114b59f19b08f838d743585bb7/en/advanced_config/parameter_reference.md)

hamishwillee, 2021-04-06:
The last change to that value was 5 years ago and the information for v1.11 and master is the same. That said, the docs are wrong because they say two things that are mutually exclusive are true.

[image]

@bkueng The name/description is "allow arming without GPS". I would therefore expect that if the value is enabled (`1`) the PX4 would allow arming without GPS and if disabled (`0`) I would expect that it would require GPS. However this is the opposite of what the value documention says. 
Further, the comment says "the default allows to arm without GPS" but the default is `1` so either the comment or the values must be wrong.

Can you clarify whether the docs for 0 and 1 are incorrect or the name/description of the param?

TaeronSystems, 2021-04-06:
The correct values are:
1: Allow arming without GPS
0: Require GPS lock to arm

What I meant in my post was that in v1.11 there is only one available interpretation, and it is correct. As you more precisely point out the information becomes self-conflicting as-of the commit identified. If the data itself were not changed, but perhaps just the way it is being displayed, this may be a more systemic issue affecting other parameters whose display style was also affected by that update.

[image]

hamishwillee, 2021-04-06:
Yes, the display changed. Thanks for pointing this out. The information is set in the source code so I have created https://github.com/PX4/PX4-Autopilot/pull/17356 on the assumption that the values are as you have indicated.

TaeronSystems, 2021-04-06:
The error rather does look like it was intentionally (if mistakenly) added on Feb 23, 2021, despite your finding that it hasn't changed in 5 years.

https://github.com/PX4/PX4-user_guide/commit/4dd371577e772e114b59f19b08f838d743585bb7

[image]

hamishwillee, 2021-04-07:
This has now updated. Thanks again for posting - much appreciated: https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_ARM_WO_GPS

closed 2021-04-07 by hamishwillee