PX4/PX4-user_guide #1320: 中文翻译和英文版图片不对应

Flordsky, 2021-06-06:
[image]
[image]

hamishwillee, 2021-06-06:
Can you post the question in English?

Ionstream, 2021-06-10:
He's trying to say that the code snippet on the mandarin and english docs do not match up.

hamishwillee, 2021-06-10:
Ah, thank you. We use crowdin to translate : http://docs.px4.io/master/zh/contribute/translation.html

Would be very pleased to accept @Flordsky into the crowdin translator group to fix it

mrpollo, 2021-06-11:
Can someone please translate the erroneous section and help us fix it? @bys1123

Flordsky, 2021-06-11:
> Ah, thank you. We use crowdin to translate : http://docs.px4.io/master/zh/contribute/translation.html
> 
> Would be very pleased to accept @Flordsky into the crowdin translator group to fix it

I had joined the group, and I will try to fix it when I was free.

Flordsky, 2021-06-11:
> Ah, thank you. We use crowdin to translate : http://docs.px4.io/master/zh/contribute/translation.html
> 
> Would be very pleased to accept @Flordsky into the crowdin translator group to fix it

I had joined the group, and I would try to fix it when I was free.

hamishwillee, 2021-06-11:
@Flordsky - when you have time is excellent. Thank you.

hamishwillee, 2021-08-19:
FYI @bys1123  Closing as this is something to deal with in Crowdin.

closed 2021-08-19 by hamishwillee