PX4/PX4-user_guide #1335: Instructions for "Writing Your First Application" outdated

Ionstream, 2021-06-10:
Followed the instruction here for `px4_simple_app`, but it will not allow you to build a working program. Code as per docs:
### Code as per docs
```
/**
 * @file px4_simple_app.c
 * Minimal application example for PX4 autopilot
 *
 * @author Example User <mail@example.com>
 */

#include <px4_platform_common/log.h>

__EXPORT int px4_simple_app_main(int argc, char *argv[]);

int px4_simple_app_main(int argc, char *argv[])
{
	PX4_INFO("Hello Sky!");
	return OK;
}
```

Here is the code as found in PX4 master branch's `/hello`:
### hello_main.cpp
```
#include "hello_example.h"

#include <px4_platform_common/app.h>
#include <px4_platform_common/init.h>
#include <stdio.h>

int PX4_MAIN(int argc, char **argv)
{
	px4::init(argc, argv, "hello");

	printf("hello\n");
	HelloExample hello;
	hello.main();

	printf("goodbye\n");
	return 0;
}

```
### hello_start.cpp
```
#include "hello_example.h"

#include <px4_platform_common/log.h>
#include <px4_platform_common/app.h>
#include <px4_platform_common/tasks.h>
#include <stdio.h>
#include <string.h>
#include <sched.h>

static int daemon_task;             /* Handle of deamon task / thread */

//using namespace px4;

extern "C" __EXPORT int hello_main(int argc, char *argv[]);
int hello_main(int argc, char *argv[])
{

	if (argc < 2) {
		PX4_WARN("usage: hello {start|stop|status}\n");
		return 1;
	}

	if (!strcmp(argv[1], "start")) {

		if (HelloExample::appState.isRunning()) {
			PX4_INFO("already running\n");
			/* this is not an error */
			return 0;
		}

		daemon_task = px4_task_spawn_cmd("hello",
						 SCHED_DEFAULT,
						 SCHED_PRIORITY_MAX - 5,
						 2000,
						 PX4_MAIN,
						 (argv) ? (char *const *)&argv[2] : (char *const *)nullptr);

		return 0;
	}

	if (!strcmp(argv[1], "stop")) {
		HelloExample::appState.requestExit();
		return 0;
	}

	if (!strcmp(argv[1], "status")) {
		if (HelloExample::appState.isRunning()) {
			PX4_INFO("is running\n");

		} else {
			PX4_INFO("not started\n");
		}

		return 0;
	}

	PX4_WARN("usage: hello_main {start|stop|status}\n");
	return 1;
}
```
### hello_example.cpp
```
#include "hello_example.h"
#include <px4_platform_common/time.h>
#include <unistd.h>
#include <stdio.h>

px4::AppState HelloExample::appState;

int HelloExample::main()
{
	appState.setRunning(true);

	int i = 0;

	while (!appState.exitRequested() && i < 5) {
		px4_sleep(2);

		printf("  Doing work...\n");
		++i;
	}

	return 0;
}
```
### hello_example.h
```
#pragma once

#include <px4_platform_common/app.h>

class HelloExample
{
public:
	HelloExample() {}

	~HelloExample() {}

	int main();

	static px4::AppState appState; /* track requests to terminate app */
};
```

hamishwillee, 2021-06-10:
Thank you. Would you like to try do a PR for an update?

Ionstream, 2021-06-10:
Give me some time to sort this out, I'm still trying to wrap my head around the starting code.

Ionstream, 2021-06-10:
If it's okay with you, I propose stretching out the "Gettting Started" section. The learning curve is rather steep, and there's quite a bit of code for new or beginning developers to take in, especially in the subsections "Writing Your First Application" and "Application/Module Template". I also feel that it's easy to get lost on certain pages, since there are too many categories and bits of information to parse through at once. Here's what I have in mind for the first part:

### 1. Writing Your First Application
#### a) "Hello Sky!" - Code
This section introduces PX4 libraries/dependencies, the `__EXPORT` keyword `PX4_INFO()` console logger,  through a simple application that prints out "Hello Sky!" and "Goodbye!" when run. It also helps to familiarise users with the structure of `PX4-Autopilot`.

#### b) "Hello Sky!" - Build
This section introduces the *build system* in PX4 through the elements `CMakeLists.txt` and `px4_add_module()` that are required compile source code into an executable to be loaded together as new firmware.  It also helps to familiarise users with the structure of `PX4-Autopilot`.

#### c) "Hello Sky" - Upload
This section introduces the tools required to flash/upload the new firmware onto PX4-supported hardware by introducing the `upload` parameter and bootloader flashing instructions (if required). It also familiarises users with the buzzer system for identifying hardware states and troubleshooting.

#### d) "Hello Sky" - Run
This section introduces the `MAVLink Shell` that can be used to interface with PX4 hardware through a command-line interface. Users will learn how to run their app/module in `nsh>` and access basic help commands.

hamishwillee, 2021-06-10:
@Ionstream Thanks very much. I mostly work on PX4 stuff on Weds/Thurs, so can't "properly" support you on this until next week. 

That said I suggest we do this in two parts. The first one being to just update this document so that it matches the current structure (but "works") and get that PR in. Then we look at restructure as a post process. 

With respect to the broader question of the getting started section it depends on what changes you want to make. The developer setup and building PX4 are "fairly" close to right in my opinion - though we should pull out everything after [troubleshooting here](http://docs.px4.io/master/en/dev_setup/building_px4.html) into its own doc. I will do that at some point.

With respect to specifically the writing application and module template, what you are suggesting sounds reasonable - and after all you are actually experiencing the problems now, so a good person to comment and attempt to fix them. In particular the module template reads as though it builds on the "writing .." doc, but feels insufficient. 

But as I say, let's just get this "not broken" first. Make sense?

Ionstream, 2021-06-11:
Sure! Acutally, there wasn't anything wrong with the code; I managed to nail it down to the "//" I used for comments, which works for C++ but not for C (noob here, sorry :P). In light of this there isn't any update required per se; apologies about that. I think we can close this issue.

That said, I'll start a PR adding some explanations/comments for the minimal application. I think it will be helpful as an entry point to inexperienced programmers like myself.

hamishwillee, 2021-06-11:
Thanks you. Really looking forward to seeing your notes.

closed 2021-06-11 by hamishwillee