PX4/PX4-user_guide #1376: External Odometry Instructions Are not Clear

aseyfi, 2021-06-23:
Hello,

I am using mocap as an external odometry source and the body frame of the mocap markers on my drone do NOT align with the drone body frame. Marker body frame is both rotated and translated in respect to drone body frame. However, I know the transformation between mocap markers body frame and drone body frame.

I was following the instructions in [external position estimation](https://docs.px4.io/master/en/ros/external_position_estimation.html) and noticed I am required to apply the transformation both in PX4 and MAVROS:

1. In [this section](https://docs.px4.io/master/en/ros/external_position_estimation.html#ekf2-tuning-configuration) I am instructed to set the following parameters in PX4:

> EKF2_EV_POS_X, EKF2_EV_POS_Y, EKF2_EV_POS_Z

Which they are 

> the position of the vision sensor (or MoCap markers) with respect to the robot's body frame.

2. Then further in the document in [this section](https://docs.px4.io/master/en/ros/external_position_estimation.html#reference-frames-and-ros) the document says:

> Your external pose system might have a completely different frame convention that does not match the one of PX4. The body frame of the external pose estimate can depend on how you set the body frame in the MOCAP software or on how you mount the VIO sensor on the drone. The MAVROS odometry plugin needs to know how the external pose's child frame is oriented with respect to either the airframe's FRD or FLU body frame known by MAVROS. You therefore have to add the external pose's body frame to the tf tree. 

Would someone please explain why do we need to set the transformation between mocap marker body frame and drone body frame in both PX4 and mavros? Why does PX4 need to know about this transformation, If mavros is capable of transforming the coordinate frames from mocap body frame to drone base_link?

Thank You,
Ahmad

hamishwillee, 2021-06-24:
@TSC21 Can you comment on this?

To me it looks wrong where we talk about setting the offsets in EKF 
> "Set the position of the vision sensor (or MoCap markers) with respect to the robot's body frame.". 

Specifically, isn't it just the vision sensor offset that you're interested in - whether it's a MOCAP or other system, this setting is just to map your sensors to the center of gravity. If I'm correct, we'd remove "or MoCap markers" from this string.

aseyfi, 2021-06-24:
@hamishwillee, @TSC21, I don't want to mix various topics here, but the other relevant problem that I see, is in the way that mavros transforms odometry data from a vision system body frame to drone center. Specifically, if there is a translation between vision system body frame (for example camera in a visual odometry system) and drone center (base_link), from rigid body dynamics we know (please see section 5.1.3 in [this link](https://www.brown.edu/Departments/Engineering/Courses/En4/notes_old/RigidKinematics/rigkin.htm)) that linear velocity of base_link will depend on the transformation between the camera and base_link and angular velocity of the camera frame. Looking at [this line](https://github.com/mavlink/mavros/blob/ros2/mavros_extras/src/plugins/odom.cpp#L247)  in mavros source code, I am not clear how this calculation could be possibly accurate without any references to odometry angular velocity.

What do you think?

hamishwillee, 2021-06-24:
I have no opinion. Normally this would be a discussion board post, that lead to documentation improvements. If you don't get answer here suggest you post on discuss.px4.io and cross link.

hamishwillee, 2021-08-19:
@TSC21 Can you comment on https://github.com/PX4/PX4-user_guide/issues/1376#issuecomment-867246727