PX4/PX4-user_guide #1388: need documentation for pixracer pro

berndpfrommer, 2021-07-05:
Now that the pixracer pro is supported, some documentation is needed.
I can volunteer to work on it, but have questions regarding the details of what sensors are being used.
According to mRo the board has 3 IMU sensors:
CIM-20602, ICM 20948, and BMI088.
Which one is used by the PX4 stack?

dagar, 2021-07-05:
> According to mRo the board has 3 IMU sensors:
> CIM-20602, ICM 20948, and BMI088.

By default there's a matrix of estimators across every combination of IMU and magnetometer (try `ekf2 status`). The best estimate is used by the system and is what sets the primary IMU for the inner loop, etc. The estimator selector will default to the first IMU unless there's a problem with it.

mRo documentation for reference - https://mrobotics.io/docs/pixracer-pro/

FYI @hamishwillee

hamishwillee, 2021-07-14:
Thanks @berndpfrommer - I've assigned to you. Is the board pixhawk-compatible or manufacturer supported? (It isn't listed in https://px4.io/autopilots/)

Assuming you don't know much about our documentation system, there is a contribution guide here: http://docs.px4.io/master/en/contribute/docs.html. Essentially it's a static site generator that takes markdown input. You can do small changes on github but for something like this you will probably need vuepress locally. 

I recommend you copy the documentation from another flight controller - perhaps http://docs.px4.io/master/en/flight_controller/pixracer.html. To make it even more useful it would be great to also have an assembly guide - that's essentially a wiring and connection guide like this one: http://docs.px4.io/master/en/assembly/quick_start_cuav_v5_plus.html

I can help you if you get stuck.

FYI, there is a little doc on IMU selection stuff in http://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#running-multiple-ekf-instances

hamishwillee, 2021-08-19:
Hi @berndpfrommer ,
Just a reminder. Do you still have bandwidth for this?

berndpfrommer, 2021-08-19:
regrettably, the answer is no: I am now living a nomadic life in a van, no longer have access to hardware, and am very busy with other projects. truely sorry!

hamishwillee, 2021-08-19:
Hi @berndpfrommer 
No worries. I'm a tiny bit jealous in my lockdown life. 
I'm going to close this - there are lots of boards that we might potentially document and there isn't much point keeping issues for specific cases without sponsors.

Happy travels!

closed 2021-08-19 by hamishwillee