PX4/PX4-user_guide #1424: Update UAVCAN Firmware Update Instructions

AlexKlimaj, 2021-07-22:
The UAVCAN firmware update instructions are out of date and don't reflect the new system that only requires the binary to be placed on the root of the sd card.

https://docs.px4.io/master/en/uavcan/node_firmware.html

hamishwillee, 2021-07-22:
@AlexKlimaj Can you create a PR please?

AlexKlimaj, 2021-08-06:
@pavel-kirienko Are the instructions still correct for the Zubax products?

pavel-kirienko, 2021-08-07:
@AlexKlimaj Zubax products (or any other UAVCAN devices) are invariant to the internals of the bootloader server. Instructions specific to Sapog are already available at https://github.com/PX4/sapog/ (and they are more up-to-date), I don't think it makes sense to keep them in the PX4 Guide. Maybe the same is true for the Thiemar ESC (does anybody use them?)

samchamberlin, 2021-09-08:
@pavel-kirienko is there a link to any documentation that Zubax would want to keep on this page?

pavel-kirienko, 2021-09-09:
There is no such documentation at the moment.

hamishwillee, 2021-09-14:
So the fix would be to  (?)
- note that firmware build process is firmware dependent, and point to each firmware instructions - e.g. Sapog: https://github.com/PX4/sapog/#firmware
- Explain where to put the files and how to start the firmware process
- Delete the rest

Correct?

hamishwillee, 2021-09-22:
@AlexKlimaj ^^^ Is this enough to information for us to move forward on this?

AlexKlimaj, 2021-09-22:
> 
> 
> @AlexKlimaj ^^^ Is this enough to information for us to move forward on this?

Yep, that should be good!

hamishwillee, 2021-09-22:
Great.  I'll leave it with you guys then.

hamishwillee, 2021-09-30:
All fixed thanks to @samchamberlin ! Thanks for keeping on top of this .

closed 2021-09-30 by hamishwillee