PX4/PX4-user_guide #1458: Offboard mode - error in the setpoint message description

squizz617, 2021-08-05:
Hi.
In `flight_modes/offboard.md`, the description of the `SET_POSITION_TARGET_LOCAL_NED` message states the following when position, velocity, and acceleration inputs are used in combination:

> the acceleration setpoint is used as feedforward; it is added to the output of the **position controller** and the result is used as the input to the **velocity controller**.

However, looking at the controller diagram (`flight_stack/controller_diagrams.md`), as well as the actual [implementation of the controller](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L126-L132), we can see that the acceleration setpoint, i.e., `_acc_sp`, is being added to the output of the PID **velocity controller**, i.e., `acc_sp_velocity`, not to the output of the position controller.

Am I correct? If so, it seems the documentation should be modified.

Thank you.

hamishwillee, 2021-08-11:
@bresch Can you please comment? Even better update the docs if incorrect :-)

hamishwillee, 2021-08-11:
FYI @bresch is on holiday. Hopefully we'll get an answer on this in a week or so.

bresch, 2021-08-19:
@squizz617 you're right, the description was not correct due to some copy/pasta mistake. I fixed it in https://github.com/PX4/PX4-user_guide/pull/1494

hamishwillee, 2021-08-19:
OK, I merged #1494. FYI though, might take a little while to go live. There are some problems with Jenkins (I think). If they don't get fixed I'll push updates live next week.

closed 2021-08-19 by hamishwillee