PX4/PX4-user_guide #1485: Mandarin docs  and crodin docs do not match up.

rzzuozuo, 2021-08-18:
Mandarin docs：https://docs.px4.io/master/zh/flight_modes/

Some additional HTML code has been added here：
[image]

[image]

The same issue appears in Korean documents

hamishwillee, 2021-08-18:
Thanks for posting. 

This is probably not going to be fixed (at least soon)
- The original source looks like this: https://raw.githubusercontent.com/PX4/PX4-user_guide/master/en/flight_modes/README.md
- The translated source looks like this: https://raw.githubusercontent.com/PX4/PX4-user_guide/master/zh/flight_modes/README.md

Both are valid HTML, but Vuepress is rendering extra stuff in the output. I have raised a bug against them here: https://github.com/vuejs/vuepress/issues/2914

This might also be done by turning the page into markdown. However that has its own problems (e.g. with bullets etc).

hamishwillee, 2025-01-29:
This should match better now. We have got rid of this HTML, which was not syncing properly.

closed 2025-01-29 by hamishwillee