PX4/PX4-user_guide #1601: This file doesn't exist https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/sitl/default.cmake

cnpcshangbo, 2021-10-17:
Hello,
https://docs.px4.io/v1.12/en/modules/hello_sky.html mentioned that we can add `examples/px4_simple_app` to a cmake file, but that cmake file doesn't exist. Maybe this link needs to be updated.
Thanks,
Bo

hamishwillee, 2021-10-17:
@cnpcshangbo Your docs link is v1.12. If you are working on v1.12 then you still need to use cmake, but all the links to the files must be from the v1.12 tree. I will fix that up. 

On master, as @dagar says you have to use kconfig.

hamishwillee, 2021-10-17:
@PetervdPerk-NXP Following kconfig is Cmakelist.txt still required?
My understanding is not, so I PR a delete here: https://github.com/PX4/PX4-Autopilot/pull/18450

For the [hellosky tutorial](https://docs.px4.io/master/en/modules/hello_sky.html#minimal-application) list item 3 adds the Cmakelists.txt and item 4 adds the kconfig file. 
- I think I need to delete the CmakeLists.txt section rigth?
- The CMakeLists point has additional info in a note about specifying a Dynamic library. Is this something we can do with kconfig and worth mentioning? If so, can you update? If not, I'll just delete the whole point.