PX4/PX4-user_guide #1675: px4-ros2 offboard control with pyrhon example

JinraeKim, 2021-11-28:
There is a cpp script for offboard control example, but it would be also helpful to provide some launchable python codes for the same example as ros2 usually supports both languages.

TSC21, 2021-11-28:
> There is a cpp script for offboard control example, but it would be also helpful to provide some launchable python codes for the same example as ros2 usually supports both languages.

@JinraeKim thanks for the suggestion. We are open for contributions in this regard from anyone willing to do them. Are you open to contribute in this regard? Thank you

JinraeKim, 2021-11-28:
> @JinraeKim thanks for the suggestion. We are open for contributions in this regard from anyone willing to do them. Are you open to contribute in this regard? Thank you

Thank you for your reply.
I'd like to do so, but I have been trying to learn how to use PX4 and ROS2; due to lack of my knowledge of both PX4 and ROS2, I don't think I can add an appropriate example. I'm really sorry.
If I find a successful example (but it could be a far future), I'll contribute.

Jaeyoung-Lim, 2021-11-28:
@JinraeKim Is there a specific reason you are trying to use python for offboard control?

hamishwillee, 2021-11-28:
I would guess the same reason everyone uses Python when they are starting - it's a much easier language to learn.

Jaeyoung-Lim, 2021-11-28:
@hamishwillee I agree, but more wondering if we should enable beginners use ROS2 + offboard.

Since the use case of ROS2 with PX4 is motivated by removing the performance bottlenecks coming from using mavlink, I was trying to understand why one would use python ROS2 with PX4. IMO, it is safer to use mavsdk-python if the developer is starting out and does not understand what is going on inside PX4 and probably ROS2 is the wrong framework to use.

hamishwillee, 2021-11-28:
@Jaeyoung-Lim Howdy! My take on this is that this is a matter of setting expectation - what can you do with each framework and programming language? What is easier for what task? What are the limits? There is no point telling a beginner "go and use MAVSDK" if that isn't going to get the performance they need for the task at hand. 

Some questions
- is there "really" a bottleneck using Python with ROS2 and offboard - note, I did not say "MAVLink"? 
- Is Python a reasonable peer of C++ in ROS.
- Where do you think the boundaries are?

If the answer is "Python is a peer of C++ for ROS" then we should try have an example of that too. And [here](http://docs.px4.io/master/en/robotics/) we should compare and contrast, with some kind of useful comparison of the options.

Can you give me some idea of where the border for usage is? Normally we say:
- "MAVSDK  has small learning curve, is intuitive, and is already customised for UAV domain". Use it where you can, which tends to be any kind of basic basic flight tasks - setting positions, taking off, landing, setting missions. 
- ROS is generic robotics. As such it has harder learning curve but it comes with a lot of useful routines for doing common stuff, and integrates well with computer vision libraries which would be hard to write yourself. So if you want to leverage existing code that would be hard to write yourself  this is worth considering.
  - Sounds like you're adding performance as a benefit here, presumably because of the ROS2 integration with uorb?

JinraeKim, 2021-11-28:
> @JinraeKim Could you maybe elaborate if there is a specific reason you are trying to use python for offboard control?

It is because I'm familiar with Python and ROS2, not with C++ (main language is Julia though).

Honestly, there are too many choices to use PX4 so I'm not sure my plan is suitable for my research.
Also, I saw in docs that the developers highly recommend that I start with ROS2 for offboard control.

JinraeKim, 2021-11-28:
> @Jaeyoung-Lim Howdy! My take on this is that this is a matter of setting expectation - what can you do with each framework and programming language? What is easier for what task? What are the [...]

A nice summary!
As a beginner of PX4, there are too many routines to perform flight control tasks, although there are less comparison between them.
IMO, there should be more comparison between many approaches in terms of 1) performance, 2) usage pattern, and 3) how easy to learn and Implement.

hamishwillee, 2021-12-01:
> Also, I saw in docs that the developers highly recommend that I start with ROS2 for offboard control.

As above, you need to start with "what you want to achieve" and that will determine the path you take. 

I have tried to integrate above summary in https://github.com/PX4/PX4-user_guide/pull/1681 . I personally don't propose to create any Python ROS examples - though if someone wants to that would be great. I consider it a minor bug that we don't.

haotianh9, 2022-01-19:
@JinraeKim I have tried to write a python version of offboard control, and it behaves the same as the original C++ version.
It is located in the scripts folder and can be ran by 
`python3 offboard_control_py.py`
I've already submitted [pull request ](https://github.com/PX4/px4_ros_com/pull/129).

hamishwillee, 2025-01-29:
I wanted to close this as stale, but I think it is still relevant to provide/link to at least one PX4 ROS2 _Python_ example.

The example provided by @haotianh9 in https://github.com/PX4/px4_ros_com/pull/129 was probably OK at the time but was never reviewed, and right now it would not fit well with the current structure of px4_ros_com.

@beniaminopozzan @Jaeyoung-Lim Can we look at https://github.com/PX4/px4_ros_com/blob/main/src/examples/offboard_py/offboard_control.py and decide if it does what we need as a basic Python example (My thinking is that it would be nice if it also emitted telemetry)?

IF the example is OK then who has Python skills to document this (I could have a go once the script is considered to deliver what a ROS/PX4 user would expect).

If no one is able/willing to get this over the line I will close this issue.

Jaeyoung-Lim, 2025-01-29:
@hamishwillee AFAIK, `px4_ros_com` should be deprecated right?

The best example I have is this: https://github.com/Jaeyoung-Lim/px4-offboard

hamishwillee, 2025-01-30:
Thansk @Jaeyoung-Lim, I have linked your repo in #3560. It would be better to have this as part of the official examples source and with documentation in the guide, but this is still way more helpful than nothing. 

> AFAIK, `px4_ros_com` should be deprecated right?

Not while we're still supporting v1.13 as it is a requirement. I have no idea what point we decide we let old versions slide.

For versions 1.14 and later it is still where we point users to for examples, so we can't shouldn't deprecate it unless that changes.

PR #3560 (2025-02-06): Offboard control Python links
Add links to examples of offboard Python.

Ideally we'd provide some worked examples in the docs for working with Python. We don't have offers to contribute those, but we do have some reasonable links to resources which we can add. If @Jaeyoung-Lim or someone else wants to add an in doc page for that, would be great, but this is better than nothing.

Fixes #1675
closed 2025-02-06 by hamishwillee