PX4/PX4-user_guide #1702: log the IMU and PWM outputs with same timestamp in pixhawk 4 mini

saitella, 2021-12-15:
Hi I am research student ,
actually to find the inertial tensor I need IMU gyro and PWM outputs values with timestamp for this how can I boost the logging data to 400HZ. The log which is generated in QGC is limited to 50HZ but I need these values at which rate the pix hawk generates the signal. so can anyone help to log the IMU gyro and PWM outputs at its maximum rate . so that it is easy for me to find the inertial tensor

hamishwillee, 2021-12-15:
That's a technical question not a bug report - you should raise this on slack.

That said, I'd be looking at http://docs.px4.io/master/en/dev_log/logging.html#logging and http://docs.px4.io/master/en/advanced_config/parameter_reference.html#SDLOG_PROFILE with bit 8 or 9 set

closed 2021-12-15 by hamishwillee

saitella, 2021-12-15:
Sorry I know that but to resolve this issue asp I am asking here.
Separately, the list of logged topics can also be customized with a file on the SD card. Create a file etc/logging/logger_topics.txt on the card with a list of topics (For SITL, it's build/px4_sitl_default/tmp/rootfs/fs/microsd/etc/logging/logger_topics.txt):

<topic_name> <interval> <instance>
The <interval> is optional, and if specified, defines the minimum interval in ms between two logged messages of this topic. If not specified, the topic is logged at full rate..
while doing this I am logging at full rate but the timestamp on the IMU data and PWM data is not same. How should I get the values with same interval of time stamp.   

thanks in advance.

hamishwillee, 2021-12-15:
Not how it works.