PX4/PX4-user_guide #1709: Unclear Fixed-wing PID Tuning Guide

danielfhnw, 2021-12-17:
While going through the [fixed-wing PID tuning guide](https://docs.px4.io/master/en/config_fw/pid_tuning_guide_fixedwing.html#tune-pitch) I was not sure, wheter I need to correctly set the pitch offset (orange box) or set it to zero (blue box).

[image]

Do I only need to set it to zero while tuning the feedforward gain and use the correct pitch offset for the P and I gains?

Am I missing something?

hamishwillee, 2021-12-21:
@danielfhnw Thanks for posting.

Firstly, if your frame allows it, we're recommending [Autotune now](http://docs.px4.io/master/en/config/autotune.html#auto-tuning)

@sfuhrer Can you please answer the question above? In summary, we set these to zero when [tuning the feedforward gain](https://docs.px4.io/master/en/config_fw/pid_tuning_guide_fixedwing.html#gains-to-set-to-zero-2): FW_PR_I, FW_PR_P, FW_PSP_OFF. BUT, do we need to keep `FW_PSP_OFF` at zero for tuning rate gain and trim offsets?

Further after we tune the rate gain, we will have tuned FW_PR_P. Do we have to set that back to zero for tuning the trim offsets?

sfuhrer, 2022-01-04:
>Do I only need to set it to zero while tuning the feedforward gain 

No I agree with you, that doesn't make sense, let's remove this line. In my eyes the PSP_OFFSET is independent of the tuning gains, so I would set it first thing and then don't though it again.

hamishwillee, 2022-01-04:
Thanks @sfuhrer - attempt to fix in #1720 but needs careful review, and possibly extra info. Generally better if you make required changes and I review - that reduces chance of me missing your main points.

PR #1720 (2022-01-06): Fixed wing manual pid - remove PSP_OFFSET
Fixes #1709

@sfuhrer I am not at all sure this is correct. Your response in https://github.com/PX4/PX4-user_guide/issues/1709#issuecomment-1004701509 was

> ... let's remove this line. In my eyes the PSP_OFFSET is independent of the tuning gains, so I would set it first thing and then don't though it again.

So, line removed. I also removed `FW_RSP_OFF ` when tuning roll.

But you said "I would set it first thing". How do we set it? This does not appear to be covered in the document?
closed 2022-01-06 by hamishwillee