PX4/PX4-user_guide #1737: Example for Mavros Custom message not working

tadteo, 2022-01-14:
The example for the creation of [Mavros Custom messages](https://docs.px4.io/master/en/ros/mavros_custom_messages.html) is not working. You can have a clearer understanding about this issue [HERE](
https://discuss.px4.io/t/error-while-data-type-conversion-from-unsigned-char-to-const-mavlink-aka-mavlink-message-type/18951/10).

hamishwillee, 2022-02-10:
@Jaeyoung-Lim @TSC21 Who would be the best person to help update this?

Jaeyoung-Lim, 2022-02-10:
@hamishwillee Thanks for bringing this up, and interestingly I wasn't even aware that such documentation existed :smile:
Currently I am falling a bit short with the availability

@tadteo If you were already trying to use this, would you be able to update the documentation? I can support you if you need any help

hamishwillee, 2022-02-15:
@tadteo Ping. Are you able to help update this a per comment ^^^ ?

tadteo, 2022-02-16:
Hi, yes sure I can try.

hamishwillee, 2023-05-24:
@tadteo Did you manage to update this?