PX4/PX4-user_guide #1739: Issue about recieving custom MAVLink messages in mavlink.md

haotianh9, 2022-01-15:
Hello PX4 developers,
    When I'm going through [mavlink.md](https://github.com/PX4/PX4-user_guide/blob/master/en/middleware/mavlink.md), I met an error 

`In file included from ../../src/modules/mavlink/mavlink_main.cpp:60:
../../src/modules/mavlink/mavlink_receiver.h:322:70: error: redeclaration of ‘uORB::Publication<ca_trajectory_s> MavlinkReceiver::_flow_pub’
  322 |  uORB::Publication<ca_trajectory_s>   _flow_pub{ORB_ID(ca_trajectory)};
      |                              `

 And I found that there is another function of `_flow_pub` defined at line 303 in [mavlink_receiver.h](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.h#L195)
`	uORB::Publication<optical_flow_s>			_flow_pub{ORB_ID(optical_flow)}; `

Thus I wonder whether there is a typo in the  [mavlink.md](https://github.com/PX4/PX4-user_guide/blob/master/en/middleware/mavlink.md). 

`uORB::Publication<ca_trajectory_s>			_flow_pub{ORB_ID(ca_trajectory)};`

Thank you!

hamishwillee, 2022-02-03:
Thanks for this. Will be looking at the example in the coming weeks.

hamishwillee, 2022-02-16:
Thanks @haotianh9 
This was fixed as you suggested in #1778. It looks like this was indeed a typo.

FYI I haven't run through the whole example, but this also now talks about the fact that now mavlink is built at PX4 build time.

closed 2022-02-16 by hamishwillee

haotianh9, 2022-02-16:
Thank you very much for fixing it!

PR #1778 (2022-02-16): Mavlink - fix typo _flow topic to _ca_traj_msg_pub
Fixes #1739

There was a typo in the Custom mavlink messages topic where the uorb topic had been renamed _ca_traj_msg_pub to _flow_pub, but one case was missed.