PX4/PX4-user_guide #1833: External Compass Orientation and CAL_MAG_ROT_AUTO

Finwood, 2022-04-04:
_Not sure if this is (also/rather) a PX4 bug, so let's start with Documentation._

During flight operations with a hexacopter running Auterion Enterprise PX4 2.4.2 (which is based on vanilla PX4 1.11.0), we have discovered undocumented behavior regarding magnetometer calibration.

Even with [`CAL_MAG_ROT_AUTO`][] set, the system seems **not** to be able to fully determine the rotation of external magnetometers, but instead only matches to the best hard-coded [orientation choice][`CAL_MAG1_ROT`] available. On our platform, we had -30° Yaw offset, which is not an available option the [`CAL_MAG_ROT_AUTO`][] algorithm is able to match. Instead of failing the calibration process and providing a meaningful error message, the compass calibration always seemed to succeed, only to start reporting errors and compass inconsistencies mid-flight.

I am unsure whether this is considered a bug in PX4 or if this behavior is intended or necessary.
At the very least, I'd like to add a paragraph in [Compass Calibration](https://docs.px4.io/master/en/config/compass.html) and a note in the description of [`CAL_MAG_ROT_AUTO`][] to prevent the next guy from weeks of debugging.

Do I see this correctly, or am I missing something? Have I missed existing documentation of [`CAL_MAG_ROT_AUTO`][]?

If you can confirm my observations, I'd be happy to create a PR to update documentation.

  [`CAL_MAG_ROT_AUTO`]: https://docs.px4.io/master/en/advanced_config/parameter_reference.html#CAL_MAG_ROT_AUTO
  [`CAL_MAG1_ROT`]: https://docs.px4.io/master/en/advanced_config/parameter_reference.html#CAL_MAG1_ROT

hamishwillee, 2022-04-06:
@bkueng The selectable `yaw`, `roll`, and `pitch` rotations are at intervals at 45 degrees. This bug reports that if you mount the compass at 30 degrees yaw and run auto calibration the calibration will "succeed", auto assigning to (presumably) the nearest selectable orientation. This is a problem because the calibration isn't actually correct and causes fails/warnings.

1. Is it that you are only _allowed_ to mount the compass in one of the supported orientations? If so, that is primarily a docs bug that I will fix.

2. Is it possible/reasonable to detect this problem during auto calibration (and fail/provide a meaningful error)?

@Finwood Thanks for posting. You're right there is a problem. Beat will be able to confirm whether it is just docs. FYI, docs on compass are (primarily): 
- https://docs.px4.io/master/en/config/compass.html
- https://docs.px4.io/master/en/gps_compass/#compass-configuration

bkueng, 2022-04-06:
> Is it that you are only allowed to mount the compass in one of the supported orientations? If so, that is primarily a docs bug that I will fix.

That's correct. At some point we'll add angle parameters to allow all custom orientations.

> Is it possible/reasonable to detect this problem during auto calibration (and fail/provide a meaningful error)?

Probably, depending on threshold and if the orientation is too close to an existing one. @dagar knows this best.

hamishwillee, 2022-04-06:
Thanks @bkueng I've created a PR for the docs and a bug report against PX4. If you and @Finwood could sanity check the docs one that would be appreciated.

PR #1835 (2022-04-07): Compass mounting/orientation doc
Calibration can "silently fail" if the orientation is set incorrectly. To make this clear I have created a new doc on mounting the compass which talks about orientation, and linked it from the calibration page. A search should at least find this now.

Fixes #1833

Note, if this is OK might then link it from the assembly pages for the different flight controllers as a post process.
closed 2022-04-07 by hamishwillee