PX4/PX4-user_guide #1849: Error in tailsitter motor numbering

zlite, 2022-04-21:
On [this page](https://docs.px4.io/v1.12/en/airframes/airframe_reference.html), the VTOL Quad Tailsitter motor numbering is wrong (and doesn't agree with the picture). Should be as shown here

[image]

hamishwillee, 2022-04-21:
Thanks very much - that's been wrong for a very long time. I have put in a fix here: https://github.com/PX4/PX4-Autopilot/pull/19518

closed 2022-04-22 by bkueng

hamishwillee, 2022-04-24:
@zlite I completely agree. When I recently updated these docs http://docs.px4.io/master/en/frames_vtol/tailsitter.html#tailsitter-airframes I originally intended to use the diagrams from airframe reference but they were too confusing.

However I don't have the SVG skills to do this (I tried, but I don't). So I won't be touching it.

zlite, 2022-04-24:
Even something as simple as extending the arm lines for the upper motors, as shown here, would give a clue. (assuming that I have indeed intuited the right orientation)
[image]

zlite, 2022-04-24:
I confirmed with Roman Bapst, the maintainer, that the above redrawing is correct

hamishwillee, 2022-04-28:
Well, I'm still not good at SVG, but why not https://github.com/PX4/PX4-user_guide/pull/1864