PX4/PX4-user_guide #1875: Beagle Bone Blue Build Error in make

Superbear88, 2022-05-02:
i stuck on line 397 with following error
Any ideas?
```
[397/960] Building CXX object src/drivers/distance_sensor/lightware_laser_ser...ers__distance_sensor__lightware_laser_serial.dir/lightware_laser_serial.cpp.
FAILED: src/drivers/distance_sensor/lightware_laser_serial/CMakeFiles/drivers__distance_sensor__lightware_laser_serial.dir/lightware_laser_serial.cpp.o
/opt/bbblue_toolchain/gcc-arm-linux-gnueabihf/gcc-linaro-12.0.1-2022.02-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DMODULE_NAME=\"lightware_laser_serial\" -DPX4_MAIN=lightware_laser_serial_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../boards/beaglebone/blue/include -I../../platforms/posix/include -Iexternal/Install/include -isystem librobotcontrol-prefix/src/librobotcontrol/library/include -O2 -g -DNDEBUG -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -Wno-cast-align -D__KERNEL__ -std=gnu++14 -MD -MT src/drivers/distance_sensor/lightware_laser_serial/CMakeFiles/drivers__distance_sensor__lightware_laser_serial.dir/lightware_laser_serial.cpp.o -MF src/drivers/distance_sensor/lightware_laser_serial/CMakeFiles/drivers__distance_sensor__lightware_laser_serial.dir/lightware_laser_serial.cpp.o.d -o src/drivers/distance_sensor/lightware_laser_serial/CMakeFiles/drivers__distance_sensor__lightware_laser_serial.dir/lightware_laser_serial.cpp.o -c ../../src/drivers/distance_sensor/lightware_laser_serial/lightware_laser_serial.cpp
../../src/drivers/distance_sensor/lightware_laser_serial/lightware_laser_serial.cpp: In member function 'virtual void LightwareLaserSerial::Run()':
../../src/drivers/distance_sensor/lightware_laser_serial/lightware_laser_serial.cpp:311:39: error: 'ssize_t write(int, const void*, size_t)' reading 5 bytes from a region of size 4 [-Werror=stringop-overread]
  311 |                         (void)!::write(_fd, &data, strlen(data));
      |                                ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[410/960] Building CXX object src/drivers/distance_sensor/teraranger/CMakeFiles/drivers__distance_sensor__teraranger.dir/TERARANGER.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:230: beaglebone_blue_default] Error 1
```

hamishwillee, 2022-05-03:
@Superbear88 I suggest you post this question on the PX4 slack or discussion boards and link back here. One option might be to remove that sensor but let's see.

Superbear88, 2022-05-05:
@hamishwillee i disabled the drivers temporaly in the board config, so i get another build error...
`[18/434] Building CXX object src/modules/control_allocator/ActuatorEffectiveness/CMakeFiles/ActuatorEffectiveness.dir/ActuatorEffectivenessRotors.cpp.o
FAILED: src/modules/control_allocator/ActuatorEffectiveness/CMakeFiles/ActuatorEffectiveness.dir/ActuatorEffectivenessRotors.cpp.o
/opt/bbblue_toolchain/gcc-arm-linux-gnueabihf/gcc-linaro-12.0.1-2022.02-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -DCONFIG_ARCH_BOARD_BEAGLEBONE_BLUE -DMODULE_NAME=\"ActuatorEffectiveness\" -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/beaglebone/blue/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../boards/beaglebone/blue/include -I../../platforms/posix/include -Iexternal/Install/include -I../../src/modules/control_allocator -I../../src/modules/control_allocator/ActuatorEffectiveness -isystem librobotcontrol-prefix/src/librobotcontrol/library/include -O2 -g -DNDEBUG -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -Wno-cast-align -O2 -std=gnu++14 -MD -MT src/modules/control_allocator/ActuatorEffectiveness/CMakeFiles/ActuatorEffectiveness.dir/ActuatorEffectivenessRotors.cpp.o -MF src/modules/control_allocator/ActuatorEffectiveness/CMakeFiles/ActuatorEffectiveness.dir/ActuatorEffectivenessRotors.cpp.o.d -o src/modules/control_allocator/ActuatorEffectiveness/CMakeFiles/ActuatorEffectiveness.dir/ActuatorEffectivenessRotors.cpp.o -c ../../src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessRotors.cpp
../../src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessRotors.cpp: In member function 'virtual void ActuatorEffectivenessRotors::updateParams()':
../../src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessRotors.cpp:98:30: error: dangling pointer to 'count' may be used [-Werror=dangling-pointer=]
   98 |         _geometry.num_rotors = count;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~
compilation terminated due to -Wfatal-errors.`

hamishwillee, 2022-05-11:
@Superbear88 As I said above, "I suggest you post this question on the PX4 slack or discussion boards and link back here."

I can't help you with these errors, because I don't know how.

hamishwillee, 2023-05-24:
Closing. Assumption is build errors were environment setup and instructions are OK.

closed 2023-05-24 by hamishwillee