PX4/PX4-user_guide #1897: Control Surfaceless VTOL Quad Tailsitters Documentational Error

aoberai, 2022-05-30:
https://docs.px4.io/master/en/frames_vtol/tailsitter.html

This page suggests that VTOL Quad Tailsitters without elevons (control surfaceless) are supported by PX4 which is not true; pitch and roll differential thrust don't exist for control surfaceless vtol tailsitters with PX4. For the PX4, elevons are mandatory and the documentation should be updated to show this.

hamishwillee, 2022-06-01:
Thanks @aoberai 

@RomanBapst Can you confirm? The specific text here says "Four rotor VTOL with optional elevons. Uses rotors to transition between modes (along with elevons if present)., which was my understanding.

How would you reword?

hamishwillee, 2022-06-02:
Or @sfuhrer Might you be able to comment?

sfuhrer, 2022-06-02:
Maybe "Four rotor VTOL with elevons for pitch/roll control in cruise flight. Uses rotors along with elevons to transition between modes."

sfuhrer, 2022-06-02:
Once we have made the small code adjustments to support elevon-less flights and tested it we can revert to the previous version.

aoberai, 2022-06-02:
Do you know what the timeline on "small code adjustments to support elevon-less flights and tested it we can revert to the previous version" would be? I would be happy to pr it in if people are busy, do you have any advice on where to look?

hamishwillee, 2022-06-09:
Thanks @sfuhrer (et al). Given you're actively looking at working on this I'd rather leave this open for a few weeks and see how it develops than update the docs :-)

hamishwillee, 2025-01-29:
Closing on the assumption this was addressed or can be re-raised if someone disagrees and thinks it is important

closed 2025-01-29 by hamishwillee