PX4/PX4-user_guide #2021: The affiliation of the catalog of books

mengchaoheng, 2022-08-29:
Why do the titles of this guide, including chapter and section titles, not explicitly indicate affiliation? The cluttered entries make it difficult for readers to read. Should this be a bad habit of programmers? ?

hamishwillee, 2022-08-29:
I have no idea what you are suggesting.

mengchaoheng, 2022-08-29:
> I have no idea what you are suggesting.

@hamishwillee When we read a book, the title is like this: it contains 1, 2 chapters, each chapter also has 1.1, 1.2 and so on. We can clearly see the structure of the whole book. I found the PX4's guide very confusing at this point. I have no idea which heading each section belongs to! ! !
For example: What is the relationship between "Homebrew Installation" and "Enforce Python Version" and "Common Tools", parallel or subordinate relationship?

[image]

mengchaoheng, 2022-08-29:
in [RTPS/DDS Interface: PX4-Fast RTPS(DDS) Bridge](https://docs.px4.io/main/en/middleware/micrortps.html), what is the relationship of Code generation , Fast RTPS(DDS) Applications (ROS Independent), Supported uORB messages, Client (PX4/PX4-Autopilot), Agent in an Offboard Fast DDS interface (ROS-independent), Creating a Fast DDS Listener application, and so on. @hamishwillee

mengchaoheng, 2022-08-29:
> I have no idea what you are suggesting.

I suggest reorganizing the title and table of contents of the guide. @hamishwillee

hamishwillee, 2022-08-29:
OK, so the problem with http://docs.px4.io/main/en/dev_setup/dev_env_mac.html is that it **does not have proper structure**. Adding section information would not fix this. I will improve that document in a few days.

For https://docs.px4.io/main/en/middleware/micrortps.html the relationships are conveyed by the size of the headings as is common in HTML. Those seem very clear to me without numbering.

mengchaoheng, 2022-08-29:
> 好的，所以http://docs.px4.io/main/en/dev_setup/dev_env_mac.html的问题是它**没有适当的结构**。添加部分信息不会解决此问题。我将在几天内改进该文件。
> 
> 对于https://docs.px4.io/main/en/middleware/micrortps.html，关系通过标题的大小来传达，这在 HTML 中很常见。这些对我来说似乎很清楚，无需编号。

Expressing relationships by title size is bullshit. You can see with the naked eye that the size of the title does not mean that other readers can. Eye disease patients with size judgment disorders will be confused by this. Please use traditional article organization instead of size. In addition, it is still difficult to discern when the difference in title size is subtle. @hamishwillee

mengchaoheng, 2022-10-11:
error
```Console
$ git clone --recursive https://github.com/eProsima/Fast-DDS.git -b v2.0.2 ~/FastDDS-2.0.2
$ cd ~/FastDDS-2.0.2
$ mkdir build && cd build
If you are on Linux, execute:

$ cmake -DTHIRDPARTY=ON -DSECURITY=ON ..
$ make -j$(nproc --all)
```
just fix by :
```Console
 git clone https://github.com/eProsima/Fast-DDS.git
git checkout v2.0.2
 git submodule update --init --recursive 
```

mengchaoheng, 2022-10-11:
@hamishwillee

closed 2023-05-24 by hamishwillee

hamishwillee, 2023-05-24:
The latest docs do not directly install this component and won't for PX4 v1.14 - so "fixed by waiting". I have tidied up the mac page so it is now less poorly structured.

The heading numbering is a "nice to have". I'm looking at updating the infrastructure again, and might add a setting to enable this for those who need it. That could be tracked in https://github.com/PX4/PX4-user_guide/issues/2542

PR #2541 (2023-05-24): Dev env mac - improve structure
The structure was confused. This partially fixes #2021 (though it will not satisfy the author).