PX4/PX4-user_guide #2116: Error in PX4-user_guide/en/ros/mavros_custom_messages.md

Edekheh, 2022-10-24:
Hi, I've been trying to compile it but there is an error in file : /PX4-Autopilot/src/modules/key_receiver/key_receiver.cpp

For compile correctly you must change lines : 
21         { .fd = keyboard_sub_fd,   .events = POLLIN },
to : 
             { .fd = key_sub_fd,   .events = POLLIN },
Then add logger library for PX4_ERR messages : 
#include <px4_platform_common/log.h>

Hope You can modify instructions or correct me if I am wrong :)

hamishwillee, 2022-10-26:
@Edekheh Thanks. what versions did you test this against? As per the document, it was tried with:

- Ubuntu: 20.04
- ROS: Noetic
- PX4 Firmware: v1.12.3

Do you think you could create a PR for the required changes? That would remove any possibility of me making a copy/paste error.

Edekheh, 2022-10-27:
@hamishwillee Sorry for late reply, i could try to create PR if You could link me to any guide/docs about how to do it properly.
And i tested in on latest 1.13 version of PX4, ROS:Noetic, Ubuntu 20.04. I think on 1.12.3 these changes are necessary too .

steamfoundry, 2022-11-14:
If your looking for some simple instructions on how to contribute, I liked the Adafruit walkthrough here. 
<https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control>

PR #2122 (2022-11-16): Pr fix custom message key receiver
PR for issue 2116 : https://github.com/PX4/PX4-user_guide/issues/2116
Added library for PX4_ERROR message and fixed variable name
Tested on 1.13, Ubuntu 20.04 and Ros Noetic

Fixes #2116
closed 2022-11-16 by hamishwillee