PX4/PX4-user_guide #2123: Pix32 V6 is not Pixhawk Autopilot Bus compatiable

double07killor, 2022-10-28:
https://docs.px4.io/main/en/flight_controller/holybro_pix32_v6.html

This page says that the Pix32 V6 complies with the pixhawk autopilot bus but it does not, it uses a different pinout as specified on the 100 pin connector link also given on the page

hamishwillee, 2022-11-03:
That says that the flight controller and the carrier board are connected by a 100 pin connector. I read this to mean that the carrier board has the Pixhawk Autopilot bus. 

@vincentpoont2 Is that correct? If so we should be more precise in the wording.

vincentpoont2, 2022-11-04:
Yea, that is a mistake and should be taken out. 
https://github.com/PX4/PX4-user_guide/pull/2131

closed 2022-11-05 by hamishwillee

double07killor, 2022-11-09:
Yeah it was a bit confusing, 

The PAB has a 100-pin and 50 pin connector, but the 50 pin is optional, which means that you could build a setup using only the 100 pin connector, but this 100 pin connector is not compatible, I would guess that it could even be a mechanically incompatible connector but at the very least the pinout is different, 

I see the page hasn't been updated yet, I only say this because I'm curious what the change will be, I almost bought a V6 to use with the PAB (I cannot find any PAB fc's in stock atm) because of that comment so I'll be happy to see when this is cleared up...

hamishwillee, 2022-11-09:
@double07killor You can see in the PR that Vince linked above: https://github.com/PX4/PX4-user_guide/pull/2131/ - he just deleted the statement it was PAB compatible.

I'll regenerate the docs on the live server later today.