PX4/PX4-user_guide #2149: autotune fixed wing QGCS tab missing

x-sven, 2022-11-19:
Hi, i would like to use autotuning for a generic fixed wing. However, using latests QGCS and PX4 firmware stack, the tab for PID tuning is missing for fixed wing configurations (VTOL or quad seems working). Documentation does not match, see: https://docs.px4.io/main/en/config/autotune.html

Vehicle setup > PID Tuning does not show up, Autotune may be enabled by flight mode selector, but unsure if that works or is desired. 
Are there any recommendations? THX Sven

Jaeyoung-Lim, 2022-11-19:
@x-sven This has been fixed in https://github.com/mavlink/qgroundcontrol/pull/10384

x-sven, 2022-11-19:
Ok, thanks for the quick reply. Can you tell me which stable QGCS version will be released with this fix?

closed 2022-11-20 by x-sven