PX4/PX4-user_guide #2186: The crypto functionality needs documentation

jlaitine, 2022-12-08:
The documentation is needed for the following

1. Signed boot
- Creating elliptic curve keys and embedding these to the bootloader (both new nuttx-based and the old one)
- Enabling signature generation at build time
- Explain how to customize signing process for production (replace cryptotools.py)
- Explain the crypto_backend architecture and how to make a custom keystore / crypto components
2. Logfile encryption
- Explain the current example implementation; xchacha20 encryption with rsa2048 encrypted key exchange
- Explain how to generate own keys for the current logfile encryption scheme (this has caused a lot of confusion due to public key part being in DER format, not raw binary). The proper way to create keys which are libtomcrypt compatible is approximately:
```
  openssl genrsa -out rsa2048.pem
  openssl rsa -in rsa2048.pem -outform der -pubout -out public.der
  xxd -i -c8 public.der

```

hamishwillee, 2022-12-08:
Sounds useful:
1. Do you have a list of associated PRs where this was contributed.
2. Can you write this for us?

jlaitine, 2022-12-08:
1. Hi, here are the most relevant PR:s
#17672 #17814 #18387 #18761 #19093 #19158 #19932 #20103

2. Yes, this was my intention, please assing the task to me. I finally started the task by creating this issue. Creating the documentation has been on my to-do list for a long time and I apologize for not being able to find time for it...

note that this is mostly documentation for developers, no so much for the end users.

hamishwillee, 2022-12-08:
Thank you - I was dreading learning enough to do a reasonable job. Yes, this should probably live somewhere around here: http://docs.px4.io/main/en/hardware/

holmnikolaj, 2022-12-08:
Great work so far @jlaitine . I can confirm the keyfile generating commands works, except the last command needs some stripping of "unsigned char public_der[] = {" and "};
unsigned int public_der_len = 294;" in top and bottom.

hamishwillee, 2023-05-24:
@jlaitine @holmnikolaj How did progress on signed boot go? Ideally this would be documented in the development section, perhaps here: http://docs.px4.io/main/en/hardware/

It's been a while, but if this is useful for other developers would be great to have.

Is there any kind of "user" story. I understand that users don't have to do anything, but it might be good to note that PX4 has this feature, the benefits, and perhaps list the constraints under which users can expect it to be working - i.e. a list of boards.

hamishwillee, 2025-01-29:
The log encryption part of this was done. Closing, with secure boot in its own issue #3553

closed 2025-01-29 by hamishwillee