PX4/PX4-user_guide #2262: Unknown Target Error when running make px4_sitl gz_x500

lmand, 2023-02-03:
The Error Message
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1

Setup
- Ubuntu: 20.04
- ROS: ROS2 foxy
- Ignition: Edifice
- latest pull from the PX4 Autopilot repo

Fixes attempted
- git clone PX4-simulation-ignition and build deb file. export IGN_GAZEBO_SYSTEM_PLUGIN_PATH=/usr/lib
- bash ./PX4-Autopilot/Tools/setup/ubuntu.sh. restart. make clean

beniaminopozzan, 2023-02-03:
If you are using the last pull, then the `gz_bridge` responsible for the connection with gazebo requires gazebo garden
```make
# Find the gz_Transport library
find_package(gz-transport
	#REQUIRED COMPONENTS core
	NAMES
		#ignition-transport11 # IGN (Fortress and earlier) no longer supported
		gz-transport12
	#QUIET
)
```
therefore you need to upgrade gazebo

and PX4-Autopilot/Tools/setup/ubuntu.sh is outdated as it installs Fortress

lmand, 2023-02-03:
Thanks. Solved. Install link if anyone needs it: https://gazebosim.org/docs/garden/install_ubuntu

closed 2023-02-03 by lmand