PX4/PX4-user_guide #2265: Update Gazebo / Ignition references

mrpollo, 2023-02-08:
The OSRF team made a few changes to Gazebo over the past months

Gazebo is now called Gazebo Classic, and the new Gazebo "Ignition" is now called "Gazebo" we need to update references on our documentation.

We should also update the OS dependency on each one and models supported

eg:

> Ubuntu 22.04 should support Gazebo Garden

hamishwillee, 2023-02-08:
I can see that initial work is happening in #2266

> Gazebo is now called Gazebo Classic, and the new Gazebo "Ignition" is now called "Gazebo" we need to update references on our documentation.

Even with your description, I find this confusing. So let's be very clear.

There is something called gazebo classic, which is the original gazebo we all know and love. 
- If I refer to `Gazebo Garden` I mean "the new version of gazebo that used to be called ignition"
- If I refer to `Gazebo Classic 9" I mean "the original version of gazebo, that used to be called gazebo 9"
- If I want to build gazebo classic, I use `make px4_sitl gazebo`
- If I want to build gazebo  I use `make px4_sitl gz`

Is that right? 

Is it just me, or is the mismatch in build vs name for gazebo classic confusing? For the love of beer, why didn't we make it?:
- If I want to build gazebo classic, I use `make px4_sitl gz_classic`

mrpollo, 2023-02-08:
IMO, we shouldn't call the new Gazebo "Gazebo Garden" only "Gazebo"

The reason for not changing the build targets is that it would break backward compatibility, which is a concern here.

You are right that Gazebo 9 is now Gazebo Classic for us.

I will leave comments on #2266 I think @junwoo091400 is also confused, which means we are not explaining this properly

hamishwillee, 2023-02-08:
This is going to be very very hard for end users, particular moving from anything before PX4 v1.13. We will have to be very careful to refer to "Gazebo Classic" everywhere. The problem will be in Ignition, where people might reasonably assume we mean gazebo classic after the change.

This requires a lot of care, which is why I would like to do the changes.

mrpollo, 2023-02-09:
I totally agree, we need to get this done very carefully and make sure its labeled properly as a "v1.14" and newer, but also making sure any classic users are aware.

Plus I haven't event touched on the new interface from PX4 to Gazebo Garden aka `gz`, which is the biggest change from Gazebo classic and PX4

beniaminopozzan, 2023-02-09:
Just to clarify for whoever reads this issue and not the related PR (where the misunderstanding is already solved)
The official target for gazebo-classic is `gazebo-classic` and the old target `gazebo` is kept to backward compatibility.

hamishwillee, 2023-02-23:
Nearly done. I think I need to check that some of the paths to simulator tools are correct.

hamishwillee, 2023-04-05:
I think this is all now done.

closed 2023-04-05 by hamishwillee