PX4/PX4-user_guide #2267: error:0308010C:digital envelope routines::unsupported on ubuntu 22.04 - Enable legacy OpenSSL provider

beniaminopozzan, 2023-02-10:
Hello, When trying to build the documentation locally on Ubuntu 22 I got this error:

```bash
pozzan@pozzan-RAIN:~/git/PX4-user_guide$ yarn docs:dev
yarn run v1.22.19
$ vuepress dev . 
wait Extracting site metadata...
tip Apply theme @vuepress/theme-default ...
tip Apply plugin container (i.e. "vuepress-plugin-container") ...
tip Apply plugin @vuepress/last-updated (i.e. "@vuepress/plugin-last-updated") ...
tip Apply plugin @vuepress/register-components (i.e. "@vuepress/plugin-register-components") ...
tip Apply plugin @vuepress/active-header-links (i.e. "@vuepress/plugin-active-header-links") ...
tip Apply plugin @vuepress/search (i.e. "@vuepress/plugin-search") ...
tip Apply plugin @vuepress/nprogress (i.e. "@vuepress/plugin-nprogress") ...
tip Apply plugin @vuepress/medium-zoom (i.e. "@vuepress/plugin-medium-zoom") ...
tip Apply plugin check-md (i.e. "vuepress-plugin-check-md") ...
tip Apply plugin mathjax (i.e. "vuepress-plugin-mathjax") ...
tip Apply plugin redirect (i.e. "vuepress-plugin-redirect") ...
tip Apply plugin dehydrate (i.e. "vuepress-plugin-dehydrate") ...
tip Apply plugin @vuepress/back-to-top (i.e. "@vuepress/plugin-back-to-top") ...
tip Apply plugin right-anchor (i.e. "vuepress-plugin-right-anchor") ...
tip Apply plugin @vuepress/google-analytics (i.e. "@vuepress/plugin-google-analytics") ...

● Client █████████████████████████ compiling (0%)  
 

ℹ ｢wds｣: Project is running at http://0.0.0.0:8080/
ℹ ｢wds｣: webpack output is served from /px4_user_guide/
ℹ ｢wds｣: Content not from webpack is served from /home/pozzan/git/PX4-user_guide/.vuepress/public
ℹ ｢wds｣: 404s will fallback to /index.html
node:internal/crypto/hash:71
  this[kHandle] = new _Hash(algorithm, xofLen);
                  ^

Error: error:0308010C:digital envelope routines::unsupported
    at new Hash (node:internal/crypto/hash:71:19)
    at Object.createHash (node:crypto:133:10)
    at module.exports (/home/pozzan/git/PX4-user_guide/node_modules/webpack/lib/util/createHash.js:135:53)
    at NormalModule._initBuildHash (/home/pozzan/git/PX4-user_guide/node_modules/webpack/lib/NormalModule.js:417:16)
    at handleParseError (/home/pozzan/git/PX4-user_guide/node_modules/webpack/lib/NormalModule.js:471:10)
    at /home/pozzan/git/PX4-user_guide/node_modules/webpack/lib/NormalModule.js:503:5
    at /home/pozzan/git/PX4-user_guide/node_modules/webpack/lib/NormalModule.js:358:12
    at /home/pozzan/git/PX4-user_guide/node_modules/loader-runner/lib/LoaderRunner.js:373:3
    at iterateNormalLoaders (/home/pozzan/git/PX4-user_guide/node_modules/loader-runner/lib/LoaderRunner.js:214:10)
    at Array.<anonymous> (/home/pozzan/git/PX4-user_guide/node_modules/loader-runner/lib/LoaderRunner.js:205:4)
    at Storage.finished (/home/pozzan/git/PX4-user_guide/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:55:16)
    at /home/pozzan/git/PX4-user_guide/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:91:9
    at /home/pozzan/git/PX4-user_guide/node_modules/graceful-fs/graceful-fs.js:123:16
    at FSReqCallback.readFileAfterClose [as oncomplete] (node:internal/fs/read_file_context:68:3) {
  opensslErrorStack: [ 'error:03000086:digital envelope routines::initialization error' ],
  library: 'digital envelope routines',
  reason: 'unsupported',
  code: 'ERR_OSSL_EVP_UNSUPPORTED'
}

Node.js v18.14.0
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

I'm using yarn version `1.22.19` and Node version `18.14.0`.

I was able to fix the issue adding
```bash
export NODE_OPTIONS=--openssl-legacy-provider
```
to my `.bashrc` following https://stackoverflow.com/questions/69692842/error-message-error0308010cdigital-envelope-routinesunsupported .

Is there a better way?

hamishwillee, 2023-02-10:
Yes, move to Windows :-). I use node v16.15.0. Does this make a difference?

beniaminopozzan, 2023-02-10:
Thanks! I'll think about that 😉

I did not try personally but version 16 should work

hamishwillee, 2023-02-15:
I did, and version 16 does work. I also updated Vuepress to the latest version and that does not help (for Node 18). It's a known issue with open ssl libraries. THere are workarounds, but just using node 16 is much easier for now. I'm going to close.

closed 2023-02-15 by hamishwillee