PX4/PX4-user_guide #2283: Gazebo iris depth camera

ViniciusAbrao, 2023-02-17:
@beniaminopozzan I can see here:

https://github.com/PX4/PX4-user_guide/blob/d88460a7594d3930f621d645b461750a29d61cad/en/simulation/gazebo_vehicles.md#quadrotor

That would have an option to make with depth camera:

make px4_sitl gazebo_iris_depth_camera

But, here in the airframes, there is not the correspondent build target:

https://github.com/PX4/PX4-Autopilot/tree/main/ROMFS/px4fmu_common/init.d-posix/airframes

Also I can see there is the model, but without the gazebo's plugin implemented in the sdf for the RGBD camera:

https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/models/iris_depth_camera/iris_depth_camera.sdf

Is the gazebo simulation with depth camera in development or should I use a different make?

ViniciusAbrao, 2023-02-17:
I could see now that the plugin is here: 

https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/models/depth_camera/depth_camera.sdf.jinja

However, actually I can not see the "libgazebo_ros_camera.so" in the following directory:

PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic

Jaeyoung-Lim, 2023-02-17:
@ViniciusAbrao This is because the ros_camera is not part of PX4, you need to install it as part of the ROS packages

ViniciusAbrao, 2023-02-17:
hi @Jaeyoung-Lim , in this case, after to install the ros_camera package, are we able to run:  

make px4_sitl gazebo_iris_depth_camera

?

Jaeyoung-Lim, 2023-02-17:
@ViniciusAbrao No, the make commands are for firmware development. Therefore it will not run ROS

ViniciusAbrao, 2023-02-17:
@Jaeyoung-Lim I mean this part of the tutorial:

https://github.com/PX4/PX4-user_guide/blob/d88460a7594d3930f621d645b461750a29d61cad/en/simulation/gazebo_vehicles.md#quadrotor

Jaeyoung-Lim, 2023-02-17:
@ViniciusAbrao So you can run the model, it will just not publish the data to ROS since ut is not being run

ViniciusAbrao, 2023-02-17:
@Jaeyoung-Lim It is not possible actually to run the gazebo_iris_depth_camera model.... 

here in the airframes, there is not the correspondent build target:

https://github.com/PX4/PX4-Autopilot/tree/main/ROMFS/px4fmu_common/init.d-posix/airframes

Jaeyoung-Lim, 2023-02-17:
@ViniciusAbrao I see, then we need to remove it from tje documentarion. Would you be able to make a fix in the docs?

ViniciusAbrao, 2023-02-17:
> @ViniciusAbrao I see, then we need to remove it from tje documentarion. Would you be able to make a fix in the docs?

Yes ... but don't you think that would be interesting to run this model in the Gazebo? 
Like we do with the "make px4_sitl gazebo" model, the only difference would be that this model also has the camera and in addition tho the topics already published by the "px4_sitl gazebo", we would have the image camera topic beeing published. Since that the plugin is in the link below, maybe it would only be necessary to change the the model when launching gazebo:

https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/models/depth_camera/depth_camera.sdf.jinja

Jaeyoung-Lim, 2023-02-17:
> Yes ... but don't you think that would be interesting to run this model in the Gazebo?

@ViniciusAbrao Not really, since the depth camera is only useful when run in the context of ROS. Otherwise it is identical to a normal `iris` model

ViniciusAbrao, 2023-02-17:
@Jaeyoung-Lim The gazebo plugin is responsible to publish to ROS topics, if the user is using ROS it can access it ...
The camera must not be published via XRCE-DDS bridge, it would only be published apart.

The need to have the
" make px4_sitl gazebo_iris_depth_camera "
running is because I can not see a different way to point the gazebo simulation for a different model. Let me illustrate like we do with the world .. if we run 
"make px4_sitl gazebo-classic_iris_vision__warehouse"
we can change the world, but as far as I understand, to change the model, would be necessary to have this 
" make px4_sitl gazebo_iris_depth_camera "
working properly.

ViniciusAbrao, 2023-02-17:
@Jaeyoung-Lim Let me put it in another way:

When we spawn a model in Gazebo, as we are doing with the make px4_sitl gazebo, we can Simulating Sensors using Gazebo.
We are with a iris  model that is fully integrated with PX4 and running in a Gazebo simulation.
Why not spawn the model including the RGBD camera? 

Only for reference, what I mean, you can see this link (where it says:"Next, let us add a depth camera to sam_bot. "):
https://navigation.ros.org/setup_guides/sensors/setup_sensors.html

If we edit the sdf file of iris model, we can simply add the RGBD plugin that publish the image in a ROS2 topic. We may do it in a different make, avoiding to change the original irisi, and calling it iris_depth_camera. In the new airframes, for the correspondent build target, could be also the from [10016_gazebo-classic_iris].

ViniciusAbrao, 2023-02-17:
> If you run with `make`, the plugin will simply not run since there is no ROS core running.

If I am not wrong, ROS2 does not need ROSCORE, the plugin will be publishing in the ROS2 topics since it starts a ROS2 node.

ViniciusAbrao, 2023-02-20:
@Jaeyoung-Lim 

I am working in a solution to spawn the iris model with rgbd in a ROS2 environment. I already developed the launch file:

https://github.com/ViniciusAbrao/px4_ros2_xrcedds/blob/master/my_package/launch/spawn_camera.launch.py

and I edited the sdf file to include the gazebo plugin:

https://github.com/ViniciusAbrao/px4_ros2_xrcedds/blob/master/my_package/models/iris/iris.sdf

They are working, so I have the drone in the Gazebo and also the camera image have been published. 

Please, do you know how to run the PX4 via ROS2 launch file?

Jaeyoung-Lim, 2023-02-20:
> If I am not wrong, ROS2 does not need ROSCORE, the plugin will be publishing in the ROS2 topics since it starts a ROS2 node.

Correct, but our discussion is about the `iris_depth_camera` which is for ROS1

ViniciusAbrao, 2023-02-20:
@Jaeyoung-Lim Maybe I had not been clear.... I intend to have not exactly the iris_depth_camera as it is today... As I told you one possible way to do it is with the ROS2 environment that I am working on. Do you know any way to run the PX4 on it?

ViniciusAbrao, 2023-02-20:
I managed to solve this problem with a different approach. It does what I am looking for, maybe in a different way I would expect, but in meanwhile it is enough. Please take a look in the section "RGBD camera simulation" of my repository:

https://github.com/ViniciusAbrao/px4_ros2_xrcedds

hamishwillee, 2023-05-24:
I don't think there is a bug here, so closing.

closed 2023-05-24 by hamishwillee

Alvin20350, 2024-06-26:
Hi, 

> I could see now that the plugin is here:
>
> https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/models/depth_camera/depth_camera.sdf.jinja
>
> However, actually I [...]

Try to run your px4 simulation like this:
VERBOSE_SIM=1 make px4_sitl gazebo-classic_iris_depth_camera
there shows more info about the errors.

In your case, this might because you forget to install some gazebo-ros libraries, 
use the command below:
sudo aptitude install ros-humble-gazebo-ros-pkgs #depend on ros2

hope this could help!