PX4/PX4-user_guide #2328: Is there anyone who can help update the video link related to the tailsitter UAV?

xdwgood, 2023-03-07:
[image]

 We recently updated the video of the AX1800 flight (including English subtitles) .
Here：

https://user-images.githubusercontent.com/24932258/223360173-80cb0d23-1719-43c3-a01c-d58ddd0a40a1.mp4

Thank you very much

hamishwillee, 2023-03-08:
Thanks @xdwgood 

All I can find on the docs so far is this [link in the gallery](http://docs.px4.io/main/en/frames_vtol/tailsitter.html#tailsitter-vtol) showing the frame - no video link :-(.

It would be great to add one. Is there any way you can upload this to youtube because that is easier for me to link to? Alternatively, if you cannot, can I have permission to upload this to the PX4 youtube account on your behalf?

Further, are there any details of your organisation you can share - i.e. links we can put below it for anyone who wants to find out more about the frame, or buy it, or similar?

hamishwillee, 2023-03-12:
Hi @xdwgood Ping ^^^

xdwgood, 2023-03-14:
@hamishwillee  Thank you for your reply. Sorry for my delay  :smiley: 

>can I have permission to upload this to the PX4 youtube account on your behalf?

Yes you can

>Further, are there any details of your organisation you can share - i.e. links we can put below it for anyone who wants to find out more about the frame, or buy it, or similar?

Our website is here: 
http://www.uav-cas.ac.cn/
http://www.laarlab.cn/
Contact Email ：tangaq.19b@igsnrr.ac.cn
Contact number：Bai Wenxing +86 13693002025 

 Thank you again for your help

hamishwillee, 2023-03-16:
Thanks very much @xdwgood 

We uploaded the video to https://www.youtube.com/watch?v=gjHj6YsxcZk and fixed up links for it in https://github.com/PX4/PX4-user_guide/pull/2347

I didn't use everyone's contact details, but I did link to http://www.uav-cas.ac.cn/ so that people can find out more information and a contact if they dig into it.

closed 2023-03-16 by hamishwillee