PX4/PX4-user_guide #2345: Final micrortps cleanup before v1.14.0 release

beniaminopozzan, 2023-03-15:
As discussed during [today's Community Q&A](https://discuss.px4.io/t/px4-community-q-a-march-15-2023/31174) in some pages the documentation is still referring to the removed `micrortps_bridge` (even if most of the transition to microdds has been completed / is happening now).

The remaining pages/sections are

- [Ethernet setup - ros section](https://docs.px4.io/main/en/advanced_config/ethernet_setup.html#ros2-setup-example). The section has already been updated and it shows that micrortps is no longer supported.
- [ROS (1) via ROS 2 Bridge](https://docs.px4.io/main/en/ros/ros1_via_ros2.html#ros-1-via-ros-2-bridge-user-guide). The page has already been updated and it shows that micrortps is no longer supported.
- [Simulation main page](https://docs.px4.io/main/en/simulation/#sitl-simulation-environment). Here the `micrortps` is just mentioned.
- [Multi vehicle simulation with Gazebo Classic](https://docs.px4.io/main/en/sim_gazebo_classic/multi_vehicle_simulation_gazebo.html). The page has already been updated and it shows that micrortps is no longer supported.

hamishwillee, 2023-03-16:
Thanks @beniaminopozzan . The comments above could be taken to mean that the needed work has been done. This is not the case - the minimal work to add a note indicating that documentation is broken has been done, but the documentation itself is still broken.

Now some of these I think we can leave in this state, but others should be easy to fix - for example you could fix https://docs.px4.io/main/en/advanced_config/ethernet_setup.html#ros2-setup-example without even testing it - mostly  just by updating the agent startup string and pointing to the existing docs.

For https://docs.px4.io/main/en/ros/ros1_via_ros2.html#ros-1-via-ros-2-bridge-user-guide Nuno suggested perhaps doing this using some new eProsima component that I can't remember the name of. That would require testing.

Were you thinking of looking at these. I do what is a priority as I get there.

Note also that I do not back fill docs. ONce we take a cut of PX4 docs that's it, the PX4 1.14 docs are more or less set in stone.

beniaminopozzan, 2023-03-16:
> The comments above could be taken to mean that the needed work has been done. This is not the case - the minimal work to add a note indicating that documentation is broken has been done, but the documentation itself is still broken.

@hamishwillee  that's sadly true, my bad choice of words.

> Were you thinking of looking at these

Yes!

> Note also that I do not back fill docs. ONce we take a cut of PX4 docs that's it, the PX4 1.14 docs are more or less set in stone.

Then I should hurry!

hamishwillee, 2023-03-16:
Thanks @beniaminopozzan .  Normally I'd say "don't worry about it, PX4 will release about a year after we say it will". But there seems to be some genuine progress happening! 

I think its time I ran a link checker again to work out what else has changed and not been documented.

mrpollo, 2023-04-04:
Hey @beniaminopozzan as discussed on today's maintainer call let's remove the ROS 1 to ROS 2 bridge page from our documentation

beniaminopozzan, 2023-04-04:
Thanks @mrpollo , here is the PR.

hamishwillee, 2023-04-04:
Did you work out what to do about the ROS2 with MAVROS story?

beniaminopozzan, 2023-04-20:
Closing as completed, all `micrortps` references have been removed.

closed 2023-04-20 by beniaminopozzan

hamishwillee, 2023-04-20:
Many thanks @beniaminopozzan