PX4/PX4-user_guide #2354: Docs sync for v1.14.0

beniaminopozzan, 2023-03-21:
As discussed in [today's Maintainers Call](https://discuss.px4.io/t/31237) some parts of the documentation are out of sync with the PX4 `main` branch. This issue aims to capture the main points that should be fixed in view of the release of `v1.14.0`

- [ ] #2345
- [ ] #2265
- [ ] [EKF2_AID_MASK](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_AID_MASK) refactoring and [ECL/EKF Overview & Tuning](https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html) update.
      Associated PRs:
      -  PX4/PX4-Autopilot#20497
      -  PX4/PX4-Autopilot#20501
      -  PX4/PX4-Autopilot#21032

If anyone is aware of other major points, please add them in the comments.

hamishwillee, 2023-03-22:
I ran a tool to find out what params have been removed and added in PX4 v1.14. The params that have been removed and are still in docs imply a docs update. The params that are new are a hint that more docs may be required. Lists are here: https://discord.com/channels/1022170275984457759/1087974671661666314

Can I get dev-team eyes on that list so we can address both cases? @mrpollo I tagged a few members, but if we could get dev team review in a call that would be good.

hamishwillee, 2023-03-22:
FYI Another task I have been working on is fixing broken links. These indicate changes that can have either small or larger impact. For example, today a broken link makes it clear that the dropix flight controller by drotek is no longer in production, so I have marked it as discontinued.

junwoo091400, 2023-03-26:
> Lists are here: https://discord.com/channels/1022170275984457759/1087974671661666314

Just a note for people viewing this issue, this list is constantly being updated in the discord, and we are keeping track of parameters left to document directly in that thread! :mechanical_arm:

mrpollo, 2023-05-03:
Can this finally be closed? https://github.com/PX4/PX4-user_guide/pull/2459

hamishwillee, 2023-05-04:
@mrpollo I was going to say "almost"

NAV_TRAFF_A_RADM and NAV_TRAFF_A_RADU were removed in https://github.com/PX4/PX4-Autopilot/pull/21283, but are still in \en\advanced_features\traffic_avoidance_utm.md and the other traffic avoidance docs.

I've asked for some support on this, but no joy. If someone has time, great, otherwise when do you think you'll really be releasing PX4 v1.14. If it is soon I'll do the work to tidy this.

junwoo091400, 2023-05-04:
> I've asked for some support on this, but no joy.

Sorry I missed this! I have pinged the author of the PR for now, but if we get no reply, I can help out!

> when do you think you'll really be releasing PX4 v1.14

From the Community QnA call [yesterday](https://discuss.px4.io/t/px4-community-q-a-may-03-2023/31965#d4-release-candidate-for-v114-15) we discussed that Release Candidate will come out by next week. However, the actual stable release schedule is uncertain 🤷

hamishwillee, 2023-05-10:
Thanks @junwoo091400 No response - did you want to look at this, or should I do it tomorrow?

junwoo091400, 2023-05-11:
> did you want to look at this, or should I do it tomorrow?

The PR was opend by the author here: https://github.com/PX4/PX4-user_guide/pull/2531

hamishwillee, 2023-05-11:
Excellent. I have reviewed. Some fixes required.

hamishwillee, 2023-05-23:
All done!

closed 2023-05-23 by hamishwillee