PX4/PX4-user_guide #2454: Axis colors for flight controller / sensor orientation docs

mortenfyhn, 2023-04-13:
Hi!

Regarding this page: https://docs.px4.io/main/en/config/flight_controller_orientation.html

and specifically these drawings:

[image]

Is there a reason why the coordinate frame axis arrows x-y-z are colored red-blue-green, respectively, instead of red-green-blue? In my experience, red-green-blue is the usual convention, possibly because it matches the common "RGB" color ordering.

hamishwillee, 2023-04-14:
No reason. I don't see it as a bug, but if someone wants to provide a version that matches then fine. However we don't have the source files, so it would be annoying pixel updates.

mortenfyhn, 2023-04-14:
I understand, thank you for the swift feedback! I might take the time to do this some day, but for now I'll close this.

closed 2023-04-14 by mortenfyhn

hamishwillee, 2023-04-14:
Makes sense!