PX4/PX4-user_guide #2550: User guide shows outdated geometry and actuator outputs section for actuator setup

idorobotics, 2023-05-27:
### Setup
OS: Ubuntu 22.04
QGroundControl: v4.2.6

### Problem
The PX4 user guide https://docs.px4.io/main/en/config/actuators.html page shows three sections for Actuator setup in QGC: Geometry, Actuator Testing and Actuator outputs. There is also an Advanced box on the top right hand corner.
[image]
 

QGC v4.2.6 only shows one section for Motor setup. All other sections are missing. There is no Advanced box shown or anyway to access the advanced options. 
[image]

### Solution
Update the documentation with the latest setup guide for Motors. 
Enable the missing section in the latest QGC if required.

hamishwillee, 2023-06-01:
@idorobotics Thank you, but this is incorrect.

The Motors and Actuator views are different. From PX4 v1.14 you'll see the first view (actuators). In PX4 v1.13 you may see either, depending on the setting of a particular parameter. The documents for PX4 v1.13 show both options, but the "main" docs reflect the actuators screen, since in main that is the only option.

closed 2023-06-01 by hamishwillee

idorobotics, 2023-06-01:
@hamishwillee thank you. How do I activate the Actuators tab on the left?  Do I need to select a specific parameter? If so, how and where? In QGroundControl v4.2.6 there is no Actuators tab, there is only Motors tab as per the screenshot.

hamishwillee, 2023-06-01:
If you install the daily QGC build and connect your flight stack and you still don't get the Actuators tab then you do NOT have the latest PX4 build. Remember this is main - which is not the stable build. After PX4 v1.13 actuators is enabled by default.

The last stable build is PX4 v1.13. That does not enable actuators by default, and it's all a bit "experimental" in that build. But if you read the docs you will find the value to set in the note near the top. https://docs.px4.io/v1.13/en/config/actuators.html#actuator-configuration-and-testing
I don't recommend it though.