PX4/PX4-user_guide #2557: Running Offboard Sensor Combined Example in Build Terminal Not Recommended

secondsensei, 2023-06-03:
ROS2 User Guide > Build ROS 2 Workspace > Running the Example says to source the `local_setup.bash` in the same terminal where `px4_msgs` and `px4_ros_com` was just built. ROS2 recommends not to source an overlay in the same terminal where it was built. ROS2 Humble documentation says it may lead to [complex issues](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html#source-the-overlay).

Jaeyoung-Lim, 2023-06-03:
@secondsensei Please consider improving the instructions by making a Pull request on the documentation

hamishwillee, 2023-06-07:
Thanks. Creating a PR is great, particularly for small changes, as you're the person who knows most about the problem. As there was no response and this seems important I have added one in https://github.com/PX4/PX4-user_guide/pull/2562

PR #2562 (2023-06-07): ROS2 - run executables in new terminal from one they were built in
Fixes #2557

This is recommendation of the ROS2 tutorial: https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html#source-the-overlay
closed 2023-06-07 by hamishwillee