PX4/PX4-user_guide #2816: Possible error in simulation instruction om v14.0

Cristian-wp, 2023-10-30:
Hello, I have recently moved from v13 to v14 and I have try to follow the model generatiom instruction to run the offboar example with ROS2.
The problem I found is at this link [https://docs.px4.io/v1.14/en/ros/ros2_comm.html#humble](url), If I use 
`make` px4_sitl gz_x500 ` 
I get in output that the gz_x500 has not been found.

If I launch:
`make px4_sitl gazebo-classic`

The model are generated correctly and Gazebo11 starts.

This is a documentation error or I am missing something?

frede791, 2023-10-30:
Have you installed gazebo? With the current version of PX4 you can only have one of gazebo-classic or gazebo, not both. Instructions on how to download Gazebo can be found here:
https://docs.px4.io/main/en/sim_gazebo_gz/

Cristian-wp, 2023-10-30:
No, I have not see that new instruction.
When I was starting to work with PX4 v13 the simulation works only on classic, thank you a lot for that link 👍🏻 

But I have another question...In the combination of ROS+Gazebo page [https://gazebosim.org/docs/garden/ros_installation#summary-of-compatible-ros-and-gazebo-combinations](url) the suggested version for ROS Humble is Fortress(and is LTS), @frede791  for your experience is mandatory to work with Garden or I can use other  versions?

frede791, 2023-10-30:
I have personally only worked with Garden (and the newer Harmonic). For those two, I have not encountered any issues so far. With the older Fortress version you may run into some compatibility issues. I would thus recommend using the newer version.

hamishwillee, 2023-10-30:
Garden is the supported Gazebo (GZ) version with PX4. I have tried Fortress and it worked, but that was some time ago.

> But I have another question...In the combination of ROS+Gazebo page [https://gazebosim.org/docs/garden/ros_installation#summary-of-compatible-ros-and-gazebo-combinations](https://github.com/PX4/PX4-user_guide/issues/url) the suggested version for ROS Humble is Fortress(and is LTS), @frede791 for your experience is mandatory to work with Garden or I can use other versions?

@beniaminopozzan I'm not quite sure how to do this, but we should note that gazebo garden is the default for use with PX4 on Ubuntu 22.04. I actually think we should probably make Ubuntu 22.04 the recommended ROS2 platform that we direct people to, and make the other ones more "hidden". Just a thought.

beniaminopozzan, 2023-10-30:
Thanks @hamishwillee for pinging me.
> I actually think we should probably make Ubuntu 22.04 the recommended ROS2 platform that we direct people to, and make the other ones more "hidden". Just a thought.

Good point, but Garden is still far from having all the features that Classic 11 has. It may be better to allow user to chose one or the other during the setup phase of PX4. My personal opinion

@Cristian-wp , you are right, the official and suggested Gazebo version for ROS Humble is Fortress.

PX4, starting from v1.14 onward, supports only Gazebo Garden and Gazebo Classic 11. On `main`, Gazebo Harmonic is supported too.

Nevertheless, Gazebo Garden & ROS Humble (and Iron, and Rolling) works just fine as long as you follow the build instructions (yeah, you have to build the packages yourself, that's the only drawback) of https://github.com/gazebosim/ros_gz and https://gazebosim.org/docs/harmonic/ros_installation

---

Regarding your initial issue, PX4 installs Gazebo Garden on Ubuntu 22.04 and Gazebo Classic 11 on Ubuntu 20.04 when you run the setup script. Updating PX4 will not automatically update your simulation environment.

Cristian-wp, 2023-10-31:
Hi @beniaminopozzan Thank you for your reply.
Just a more question... for your personal experience is better develop a PX4 based project with Gazebo11 or Garden/Harmonic for the simulation part? In a few months I will start a new project, and I don't want to develop something that is already dated.
I think that Gazebo11 as you say have more feature at the moment(for example PX4 have already a model with a gimbal), but Garden and similar are becoming the standard for ROS2 simulation.

beniaminopozzan, 2023-11-01:
In my opinion, I would check if Garden/Harmonic already satisfies all my requirements, and if it does not, how difficult it is to fix those issues.
I'll try to use the new version as baseline though.

What do you want to do exactly?

Cristian-wp, 2023-11-06:
I would like to perform some indoor test with slam using 3D lidar, depth cameras, laser distance sensors and use a camera vith gimball to take photos of the simulation enviroment.

hamishwillee, 2023-11-08:
> I would like to perform some indoor test with slam using 3D lidar, depth cameras, laser distance sensors and use a camera vith gimball to take photos of the simulation environment.

Then right now you will have to use Gazebo Classic in Ubuntu 20.04, because the Gazebo does not appear to yet support camera integration etc.

Cristian-wp, 2023-11-08:
@hamishwillee I see that, but as you know, is possible use the Typhoon in Gz Garden in some way?

frede791, 2023-11-08:
I have not encountered this particular issue but I have experience some troubles with Harmonic over the past couple of days. Best stick with Garden for now.

hamishwillee, 2023-11-15:
> @hamishwillee I see that, but as you know, is possible use the Typhoon in Gz Garden in some way?

I do not know. But I suspect "not supported yet". We'd welcome the contributions - eventually we want to drop classic altogether but Gz needs to be a reasonable alternative before that could happen.

frede791, 2023-11-15:
Correct. The typhoon is not yet supported and a bunch of plugins necessary to theoretically port it, primarily camera related, are not yet available in Ignition Gazebo

hamishwillee, 2024-06-05:
OK. Closing.

closed 2024-06-05 by hamishwillee