PX4/PX4-user_guide #2831: I ran into a problem running the command px4_sitl jmavsim

zhangshuaiqing, 2023-11-01:
when I run command 
make px4_sitl jmavsim
then return 
-- PX4 version: 4b99bd2077
-- PX4 config file: /home/zsq/package/PX4_about/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
Checked src/lib/ecl submodule, ACTION REQUIRED:

Different commits:
* src/lib/ecl b3c1bd6...5d34d7a:
  Warn: src/lib/ecl doesn't contain commit b3c1bd6d40952a9d467497162f91f28400ae0a3d

 *******************************************************************************
 *   IF YOU DID NOT CHANGE THIS FILE (OR YOU DON'T KNOW WHAT A SUBMODULE IS):  *
 *   Hit 'u' and <ENTER> to update ALL submodules and resolve this.            *
 *   (performs git submodule sync --recursive                                  *
 *    and git submodule update --init --recursive )                            *
 *******************************************************************************

   Only for EXPERTS:
   src/lib/ecl submodule is not in the recommended version.
   Hit 'y' and <ENTER> to continue the build with this version. Hit <ENTER> to resolve manually.
   Use git add src/lib/ecl && git commit -m 'Updated src/lib/ecl' to choose this version (careful!)

y
Continuing build with manually overridden submodule..
Checked src/lib/matrix submodule, ACTION REQUIRED:

Different commits:
* src/lib/matrix 3679f7f...c8da796:
  Warn: src/lib/matrix doesn't contain commit 3679f7fd5160c7d2a8271b1134ca1e601f1a4254

 *******************************************************************************
 *   IF YOU DID NOT CHANGE THIS FILE (OR YOU DON'T KNOW WHAT A SUBMODULE IS):  *
 *   Hit 'u' and <ENTER> to update ALL submodules and resolve this.            *
 *   (performs git submodule sync --recursive                                  *
 *    and git submodule update --init --recursive )                            *
 *******************************************************************************

   Only for EXPERTS:
   src/lib/matrix submodule is not in the recommended version.
   Hit 'y' and <ENTER> to continue the build with this version. Hit <ENTER> to resolve manually.
   Use git add src/lib/matrix && git commit -m 'Updated src/lib/matrix' to choose this version (careful!)

y
Continuing build with manually overridden submodule..
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library 5d34d7a
[... 88 lines cut ...]
   src/drivers/gps/devices submodule is not in the recommended version.
   Hit 'y' and <ENTER> to continue the build with this version. Hit <ENTER> to resolve manually.
   Use git add src/drivers/gps/devices && git commit -m 'Updated src/drivers/gps/devices' to choose this version (careful!)

y
Continuing build with manually overridden submodule..
[11/809] Generating git version header
FAILED: src/lib/version/build_git_version.h 
cd /home/zsq/package/PX4_about/PX4-Autopilot && /usr/bin/python3 /home/zsq/package/PX4_about/PX4-Autopilot/src/lib/version/px_update_git_header.py /home/zsq/package/PX4_about/PX4-Autopilot/build/px4_sitl_default/src/lib/version/build_git_version.h --validate

Error: the git tag '4b99bd2077' does not match the expected format.

The expected format is 'v<PX4 version>[-<custom version>]'
  <PX4 version>: v<major>.<minor>.<patch>[-rc<rc>|-beta<beta>|-alpha<alpha>|-dev]
  <custom version>: <major>.<minor>.<patch>[-rc<rc>|-beta<beta>|-alpha<alpha>|-dev]
Examples:
  v1.9.0-rc3 (preferred)
  v1.9.0-beta1
  v1.9.0-1.0.0
  v1.9.0-1.0.0-alpha2
See also https://dev.px4.io/master/en/setup/building_px4.html#firmware_version

[16/809] Generating uORB topic headers
ninja: build stopped: subcommand failed.
make: *** [Makefile:228: px4_sitl] Error 1

zhangshuaiqing, 2023-11-01:
what should I do to fix this problem

hamishwillee, 2023-11-02:
This happens because you have submodules that don't match the precise build of PX4 that you are using. YOu may have changed branches or similar. THe fix is described here: http://localhost:8080/px4_user_guide/en/contribute/git_examples.html#changing-source-trees

Essentially you need to clean your environment:
```
make distclean
make clean
make submodulesclean
```

Then you should be able to build without all those errors.

closed 2023-11-02 by hamishwillee